guysoft/OctoPi #228: Trying to install vncserver on a Pi3 with octoprint installed and failing.

gzcwnk, 2016-05-13:
When I login via vncviewer (via ssh tunnel) I get,

Xsession: unable to start X session --- no "/home/pi/.xsession" file, no 
"/home/pi/.Xsession" file, no session managers, no window managers, and no 
terminal emulators found; aborting

Now I have a pi2 running vncserver and raspberrian no issues.

markwal, 2016-05-13:
octopi, thankfully, is now made with a Jessie Lite image that does not include XWindows or a GUI, so it is much faster to download and image onto the SD card.  You can install all the prereqs for that yourself though via apt-get install.

gzcwnk, 2016-05-15:
Hardly over-powering reasons, maybe saving 2~3mins, and 1 or 2 gb of space, but thanks at least I know why it now doesnt work.    Now I have to find how to upgrade it to a usable state which will way waste more time than I saved downloading and DD ing it.

foosel, 2016-05-16:
Quoting the release notes:

> If you really need a desktop environment on your OctoPi instance (note that you do not need a desktop for regular operations!), you can however install it yourself using the bundled installer script: sudo ~pi/scripts/install-desktop

And about this:

> maybe saving 2~3mins, and 1 or 2 gb of space [...] which will way waste more time than I saved downloading and DD ingteh smaller image.

VNC and X are not needed for OctoPi's core use case. Forcing thousands of people to download a twice as large image (and us to pay the traffic for that) just so a handful of people who still want that stuff included can save the couple of minutes it requires them to install that stuff on demand does hardly make sense. And in fact we got a lot of complains about the size of the image before - not everyone has fast and reliable Internet. 

You are welcome to build your own version of OctoPi based on the full raspbian image and host it.

closed 2016-05-16 by foosel