guysoft/OctoPi #229: Updating system breaks network configuration (wireless settings set via octopi-network.txt)

nicolinux, 2016-05-15:
I just "apt-get update; apt-get upgrade". After a reboot the network configuration seems to be broken. The script /root/bin/map_iface always returns "-raspbian" even though the /boot/octopi-network.txt was not changed.

Octopi 0.13.0

foosel, 2016-05-15:
The map-iface return value is probably not the reason (you need to call it with the interface as parameter and it can't properly match a config without that). 

My guess is that the special Raspbian network package once again decided to just blindly replace the existing /etc/network/interfaces file instead of keeping its hands off of that. If so there's nothing that can be done about that for 0.13 from our side. An upgrade should not nuke a critical system configuration file, but it would not be the first time that happens with that stuff sadly. 

Going forward I'd actually vote for removing all this custom raspbian only network stunts they pulled (it's all done by one specific package, including this horrible self built DHCP client setup) and have a proper pure /etc/network/interfaces that only looks at the octopi-network.txt file, instead of bending over backwards left and right trying to stay compatible to that raspbian only non standard crud and then getting the floor pulled out underneath regardless.

foosel, 2016-05-17:
So I just got around to trying that myself. Used an OctoPi 0.13.0 image. Did `sudo apt-get update` followed by `sudo apt-get upgrade`. Rebooted. Still everything working as it should.

Can't reproduce, need more instructions :)

nicolinux, 2016-05-17:
Hm, I can't say that I am very motivated to try this again because it means that I'll have to dig around for the Raspberry Pi and rip out cables from elsewhere in order to connect it to a keyboard and monitor.
Let's close this issue for now then.
And sorry if I caused you trouble/additional work.

closed 2016-05-17 by nicolinux