guysoft/OctoPi #230: Flashing latest img to SD, no response from Pi 2

GottaBeKD182, 2016-05-15:
I apologize if this is the wrong place to post this, but I've recently gotten a Pi 2, and have followed Tom's guide on how to install OctoPi, but seem to be running into an issue. After changing the network settings, plugging the SD into the Pi and plugging in the power, nothing happens. The red and green lights are on, but the CanaKit WiFi dongle doesn't light up, I'm guessing meaning there's an issue with my network settings.

I've installed the basic OS that came with the Pi and it can access the network just fine that way, so I know it's not a faulty dongle. I've also tried re-flashing the image, and entering my network info again, but no change. 

Is there something else I can do to get this working? Thanks for the advice, I really appreciate it.

markwal, 2016-05-16:
Your diagnosis is correct. Your Network settings are not correct.  You'll need to be more specific about how you set then for us to help. Which file did you edit, using what editor and what lines, specifically, did you change (including removing the comment marker #)?

GottaBeKD182, 2016-05-16:
Okay so here's some specifics:

I edited the octopi-network.txt file with Notepad++
I removed the comment marker from lines 14, 15, and 16, and typed in the name of my network in line 15, along with the password in line 16, without removing the quotations on either. Once that didn't work, I did the exact same thing with lines 19, 20, and 21 just in case that was the setup for my network.
I saved the document, and safely removed the SD card, plugged it into the SD slot on the Pi, and plugged in the power cord.

If you need any other info please let me know. Thanks so much for the help, I really appreciate it.

markwal, 2016-05-16:
Does your ssid or password have quotation marks or spaces?

You may need to wire it with Ethernet so you can ssh in to do some spelunking with the command line to look at logs, iwlist, ifup, etc.

pcbroch, 2016-05-16:
If you have an HDMI cable and HDMI enabled TV, try and plug it in to see what happens at boot up on the console.  I was going crazy thinking my network wasn't working, but it turned out to be a bad SD Card that caused the boot to hang halfway through, never reaching the network  part.

GottaBeKD182, 2016-05-17:
@pcbroch That's essentially what I've done from the start. When I first got the Pi I installed the OS that came with it, and had my TV connected so I could make sure everything was working right. When I changed the SD card to the one with OctoPi on it, the HDMI cable was still connected like it was when I was using the Pi's OS, and nothing shows up on screen. The SD I'm using for OctoPi is a spare one I had, whereas the one with the Pi's OS was the one that came with my CanaKit set. Maybe I'll backup the files from the new card and try flashing to that SD instead...

@markwal The SSID I have is all one word, no numbers, spaces or quotation marks. The password is all numbers.

markwal, 2016-05-17:
The image sounds broken, what method did you use to make the image.

pcbroch, 2016-05-17:
I agree with @markwal   If nothing shows up on the screen, you have a problem earlier than the network initialization.  I'd swap the cards or try a different method to burn the image.  I've had good luck with "ApplePi Baker" on the mac.

GottaBeKD182, 2016-05-17:
@markwal I followed Tom's Guide that's on the download page of the OctoPi site. I used Win32DiskImager. Specifically I opened the program, selected the .img file, made sure drive F was selected (which definitely was my SD card) and hit "Write". I safely removed the SD before putting it into the Pi 2.

@pcbroch I'm running Windows 10, any Windows options I could try in case switching SD cards doesn't work?

pcbroch, 2016-05-19:
@GottaBeKD182 , did you resolve your issue?  I'm sorry, I don't have any suggestion for Windows as I don't use it for these types of things.

GottaBeKD182, 2016-05-21:
@pcbroch Yes I did... Kind of. Swapping SD cards fixed the issue, and now OctoPrint is running as it should... Aside from one issue. When I tell it to connect to my printer, it says "Detecting Baudrate", displays that for a minute or two, then gives me an error of: "Unhandled Communication Error: The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: No more baudrates to test, and no suitable baudrate found." While on the printer itself the bottom line reads "Wait for user..."

So again it looks like I'm stuck. Is this a common error?

markwal, 2016-05-21:
@GottaBeKD182 probably all you have to do now is choose a specific baud rate (115200 or 250000) rather than auto.

pcbroch, 2016-05-21:
Was going to say the same thing.  For some reason, Auto does not work for me either.  I have a Printrbot and specifying 250000 allows me to connect no problems.

foosel, 2017-02-28:
Original issue reported as fixed, and follow up issue is not an issue with OctoPi but sounds rather like a support issue best tackled via OctoPrint's [mailing list](https://groups.google.com/group/octoprint) or [the G+ community](https://plus.google.com/communities/102771308349328485741).

closed 2017-02-28 by foosel