guysoft/OctoPi #231: SSHD is down by default

orcinus, 2016-05-29:
Flashed latest available image, the Pi is available via bonjour and IP, octoprint is up and accessible via http and https, however, SSHD seems down as port 22 is closed. Which is kind of a nag, considering my Pi is buried inside the printer at the moment.

nmap output:

PORT     STATE SERVICE
80/tcp   open  http
443/tcp  open  https
8080/tcp open  http-proxy

Why is SSHD not configured to start automatically on first boot anymore?

orcinus, 2016-05-30:
Bizarre.
Two reboots later, it finally started.

PORT     STATE SERVICE
22/tcp   open  ssh
80/tcp   open  http
443/tcp  open  https
8080/tcp open  http-proxy

Any ideas on what happened? Is there some kind of a delayed start? Corrupt image maybe?

pcbroch, 2016-05-30:
Impatient user?  Upon first boot, it needs to generate the keys, which
takes a while.  I don't know how long you waited but it may be that you
simply didn't wait long enough?

orcinus, 2016-05-30:
My RP3 is being extremely slow for some reason (crappy SD card, maybe) so it's possible.

I waited long enough to find the right IP (took a while for bonjour to kick in), check if everything works correctly in OctoPrint, whether the camera is functional and i can connect to the printer, update OctoPrint (which failed restarting afterwards for some reason, with a not very useful log message), then waited for some 10-15 minutes afterwards.

On the next 2 boots i did wait a pretty short time, though.

orcinus, 2016-05-30:
Anyway, closing, whatever it was, obviously not related to OctoPi.

closed 2016-05-30 by orcinus

orcinus, 2016-05-30:
Just FYI, in case anyone comes across this, it appears to have indeed been due to a very slow microSD card (class 4). Tried with a sandisk class 10 card, SSHD fired up immediately after boot.