guysoft/OctoPi #232: Armbian support for OctoPi

guysoft, 2016-06-04:
Hey all,
I have opened a new  [branch called ``armbian``](https://github.com/guysoft/OctoPi/tree/armbian), which has a variant under that name that lets you build OctoPi on [Armbian](http://www.armbian.com/). There is a [a big list of devices that run armbian](http://www.armbian.com/download/). And it would be interesting to hear if anyone there is a call for a popular device.

For now I am testing using a HummingBoard kindly provided by SolidRun, and it seems like OctoPrint, wifi system and webcams work smoothly.
The beta image of the HummingBoard is available here:
https://storage.googleapis.com/octoprint/Armbian_5.10_Cubox-i_Debian_jessie_3.14.67-0.14.0_beta.zip

md5:
https://storage.googleapis.com/octoprint/Armbian_5.10_Cubox-i_Debian_jessie_3.14.67-0.14.0_beta.zip.md5

universam1, 2016-06-15:
Seriously interested for the Orange PI PC.
Will try this branch, thanks for good work!

guysoft, 2016-06-15:
1. Download the armbian image you want, and place it in ``src/image-variants``
2. You will need to 7z extract it, and then zip (I just didn't write a method that can work with 7z yet). Note the zip should start with the word ``Armbian`` and end with ``.zip``.

3. Then run
```
git checkout armbian
cd src
sudo ./build armbian
```
or if you using the vagrant method:

```
git checkout armbian
cd src/vagrant
sudo ./run_vagrant_build.sh armbian
```
Like any other variant.

To confirm the build is indeed using the correct variant you can see in the config setting its picking the Armbian image, here is an example output:
```
--> Building VARIANT armbian, FLAVOR default
Sourcing variant config /OctoPi/src/variants/armbian/config...
================================================================
Using the following config:
OCTOPI_APT_CACHE = /OctoPi/src/workspace-armbian/aptcache
OCTOPI_APT_PROXY = 
OCTOPI_BUILD_REPO_MIRROR = 
OCTOPI_CHROOT_SCRIPT_PATH = /OctoPi/src/chroot_script
OCTOPI_COMMIT = 11f143544c51e016d20bb469390a9856a01a365c
OCTOPI_CURAENGINE_ARCHIVE = https://github.com/Ultimaker/CuraEngine/archive/15.04.6.zip
OCTOPI_CURAENGINE_VERSION = 15.04.6
OCTOPI_FBCP_REPO_BRANCH = 
OCTOPI_FBCP_REPO_BUILD = https://github.com/tasanakorn/rpi-fbcp
OCTOPI_FBCP_REPO_DEPTH = 1
OCTOPI_FBCP_REPO_SHIP = https://github.com/tasanakorn/rpi-fbcp
OCTOPI_IMAGE_ENLARGEROOT = 400
OCTOPI_IMAGE_PATH = /OctoPi/src/image
OCTOPI_IMAGE_RASPBIAN = no
OCTOPI_IMAGE_RESIZEROOT = 500
OCTOPI_INCLUDE_CURAENGINE = yes
OCTOPI_INCLUDE_HAPROXY = yes
OCTOPI_INCLUDE_MJPGSTREAMER = yes
OCTOPI_INCLUDE_OCTOPIPANEL = no
OCTOPI_INCLUDE_OCTOPRINT = yes
OCTOPI_INCLUDE_WIRINGPI = no
OCTOPI_MJPGSTREAMER_REPO_BRANCH = master
OCTOPI_MJPGSTREAMER_REPO_BUILD = 
OCTOPI_MJPGSTREAMER_REPO_DEPTH = 1
OCTOPI_MJPGSTREAMER_REPO_SHIP = https://github.com/jacksonliam/mjpg-streamer.git
OCTOPI_MOUNT_PATH = /OctoPi/src/workspace-armbian/mount
OCTOPI_OCTOPIPANEL_REPO_BRANCH = master
OCTOPI_OCTOPIPANEL_REPO_BUILD = 
OCTOPI_OCTOPIPANEL_REPO_DEPTH = 1
OCTOPI_OCTOPIPANEL_REPO_SHIP = https://github.com/jonaslorander/OctoPiPanel.git
OCTOPI_OCTOPRINT_REPO_BRANCH = master
OCTOPI_OCTOPRINT_REPO_BUILD = 
OCTOPI_OCTOPRINT_REPO_SHIP = https://github.com/foosel/OctoPrint.git
OCTOPI_OVERRIDE_HOSTNAME = octopi
OCTOPI_PATH = /OctoPi/src
OCTOPI_POSTSCRIPT = 
OCTOPI_PRESCRIPT = 
OCTOPI_PYPI_INDEX = 
OCTOPI_ROOT_PARTITION = 1
OCTOPI_SCRIPT_PATH = /OctoPi/src
OCTOPI_WORKSPACE = /OctoPi/src/workspace-armbian
OCTOPI_ZIP_IMG = /OctoPi/src/image-variants/Armbian_5.10_Cubox-i_Debian_jessie_3.14.67.zip
================================================================
/OctoPi/src/workspace-armbian /OctoPi
Archive:  /OctoPi/src/image-variants/Armbian_5.10_Cubox-i_Debian_jessie_3.14.67.zip
  inflating: Armbian_5.10_Cubox-i_Debian_jessie_3.14.67.raw 
```

You can see its extracting Armbian_5.10_Cubox-i_Debian_jessie_3.14.67.raw. 
If you want to re-run it, delete the folder ``src/workspace-armbian``

guysoft, 2016-06-26:
@universam1 Any luck, I was considering building it for you on condition you test it and report back here

universam1, 2016-07-05:
@guysoft Sorry for late reply - I was struggling with Wifi driver on Armbian.

I am very willing to test, thanks for help!

universam1, 2016-07-05:
Having following issues when running the build under Ubuntu 14.04 VM:

```
++ this_command='apt-get update'
+ apt-get update
Err:1 http://ports.ubuntu.com xenial InRelease
  Could not resolve 'ports.ubuntu.com'
...
E: Failed to fetch http://ports.ubuntu.com/pool/main/a/....
```
... and so on, fails at the end.

Looks like the mounted image cant reach out to internet, although the URLs that fail all are resolvable if tried manually from within the VM.

Any ideas?

guysoft, 2016-07-05:
@universam1 Here, have fun and report back:

https://storage.googleapis.com/octoprint/Armbian_5.14_Orangepipc_Debian_jessie_3.4.112-0.14.0_beta.zip
md5:
https://storage.googleapis.com/octoprint/Armbian_5.14_Orangepipc_Debian_jessie_3.4.112-0.14.0_beta.zip.md5

guysoft, 2016-07-05:
Can you send pull requests for those changes?

tedo2007, 2016-09-10:
Any news on this?

guysoft, 2016-09-10:
As far as I know it works. Just pick a device and build for it. There are.
40+ devices supported so I cannot build for them all.

universam1, 2016-09-11:
Yes it runs stable since weeks! No issues, thanks!

tedo2007, 2016-09-12:
Thanks.  Universam1 what device did you use i have a orange pi lite

tedo2007, 2016-09-12:
Also how did you go about installing?

universam1, 2016-09-12:
I use a orange pi pc with 1gb ram. That's never needed, 512 will be way
enough.

guysoft, 2016-09-14:
Branch has been merged In case anyone missed that

tedo2007, 2016-09-16:
Sorry for the dumb question but what does that mean?

guysoft, 2016-09-17:
Can you be more specific - what do you mean by "that"?

tedo2007, 2016-09-17:
Sorry, what does merged mean when it comes to the branch. I know what branching means.

tedo2007, 2016-09-17:
Disregard. Figured it out. Thanks

poorredneck, 2017-02-12:
I hate to be "that guy" but I have been struggling to get OctoPrint running on Armbian 5.25 when installed manually. Won't autostart on boot, and doesn't recognize USB port. 

I am very new to linux, and have no idea how to build one from here. I have the Armbian 5.25 Desktop version with Ubuntu Xenial image, how do I go about making this for the Orange Pi Lite using Armbian 5.25 and Octoprint?

I don't have a computer that runs linux, I guess I could try and get a VM going. Could it be built on the Orange Pi? 

Thank you in advance for any help.

anthonywob, 2017-02-12:
There is a good guide here  http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=594 That goes through the steps. I've done it but couldn't get autostart as well so just use Terminal to start it and run int he background then close the shell and leave it be. I'm a noon to Linux as well so learning as I go as well.

guysoft, 2017-02-12:
@anthonywob OctoPi can be built for Armbian. I think knowing how to build is much better, it lets you switch the base image when a new image is out. Long winding guides is why I built OctoPi.

@poorredneck Can you please link the exact base image you want?
There is already an OrangePi build at the top of this issue here: https://github.com/guysoft/OctoPi/issues/232#issuecomment-230454412
It could help if you explain why you need a different one.

poorredneck, 2017-02-13:
Armbian 5.25 Desktop
Direct Download Link: 
https://dl.armbian.com/orangepilite/Ubuntu_xenial_default_desktop.7z

My issue is more of ignorance than anything lol. This Orange Pi is the first time I've touched linux in 10 years. I dabbled with it in high school but most of that knowledge is gone. 
Is the version linked above a "server" version (no desktop GUI) I would like to retain the desktop GUI as I will eventually be using this PI for not just Octoprint, and would just like to have a GUI environment. 

I am running that version of Armbian right now, with Octoprint installed manually but as stated above can not get it to auto boot, or recognize USB/Serial port. I was using a very old image before hand but CPU scaling was broken and was stuck at min frequency no matter what I tried to fix it. Because it is so old, no-body would support it or assist with debugging it. Everybody says to start fresh. I did just that but now I have no working Octoprint for the reasons listed above.

Thank you for the fast response!

poorredneck, 2017-02-13:
@anthonywob that is exactly the same article that I used. My issue is when I close the terminal shell the instance of Octoprint stops as well. 
Also there is not USB/Serial. Only option on the drop down is Auto and that doesn't work.

poorredneck, 2017-02-13:
So just to make sure I understand this correctly.
Presumption #1: I can build this on the Orange Pi
Presumption #2: I can download any version of Armbian, just copy it to the /img/src directory.

The tutorial is assuming I have more knowledge than I really do. I can usually figure stuff out, but I'm just having difficulty with this.

The 7z that is downloaded will contain a .img file. I am to then ZIP that .img file with the proper naming structure outlined above. Armbian.zip would work.
I then copy that file to /img/src/

The next step you say : git checkout armbian

Is this assuming I have cloned it to the OrangePi first? 
$ git -clone https://github.com/guysoft/OctoPi.git

poorredneck, 2017-02-13:
I'm trying the 5.14 Image file you have linked above. Completely overlooked that part!

poorredneck, 2017-02-13:
Ok, so I have burned and setup the image listed above... but how the hell do I connect to WiFi?
nmtui doesn't work on the command line. Tried running sudo nmtui and says not found.
Can't edit the network.txt file on the boot partition. 

Would it be possible for somebody to build a version using the image here:
https://dl.armbian.com/orangepilite/Ubuntu_xenial_default_desktop.7z

anthonywob, 2017-02-13:
I don't know enough to help you out with the process I used.
Start the Octoprint server ./run
Then press ctrl + z   "this stops/pauses the process"
Then type bg   "moves the process to the background"
Then type disown -h  "makes it so it doesn't end when you close the shell"

poorredneck, 2017-02-13:
I'm going to attempt to build one with the image of my choice. Does this have to be done on the Orange Pi or can I use a linux VM? I've downloaded Ubuntu 16.10 on a virtual box.

poorredneck, 2017-02-13:
Damnit, I can't even unzip the freakin image file without doing something wrong! I get a screen full of errors with access denied, then prompts to over write existing files that are different sizes. There is no existing folder there that I'm extracting to so I don't understand how there could be over write prompts. 

I am supposed to use 7zip to unzip the .img file correct? 
[image]

I am willing to paypal somebody $$$ to just please just build an octoprint image file using the armbian 5.25 desktop for the orange pi lite. I'm so fed up with getting close enough to smell it and then something not working.

guysoft, 2017-02-13:
@poorredneck Thats a full desktop image, why not use https://www.armbian.com/donate/?f=https://dl.armbian.com/orangepilite/Ubuntu_xenial_default.7z ?
Or better still the debian build at:
https://www.armbian.com/donate/?f=https://dl.armbian.com/orangepilite/Debian_jessie_default.7z

Also check out the known issues, expect wifi problems with the built in wifi:
https://www.armbian.com/orange-pi-lite/

Tell me which one you want and I can build it for you.

poorredneck, 2017-02-13:
Hi Guysoft, thank you very much for your help. It is sincerely appreciated.

I would like to use the one with full desktop image. I feel alot more comfortable with having a GUI, and I have a plan for using the HDMI out with a 7" LCD to display the web interface on the 3D Printer, using the Pi as both the octoprint server and to display the web page. I have tried touch UI, and didn't care for it.  

https://dl.armbian.com/orangepilite/Ubuntu_xenial_default_desktop.7z

Thank you again for your help!

poorredneck, 2017-02-15:
I figured I'd post this to help any others in case they come across this. I finally got octoprint to start on its own with Armbian 5.25 desktop.

My OrangePi is setup so the "default" user (the user that logs into the desktop environment on startup) is octoprint. The user that runs the OctoPrint software is "pi".

I followed the instructions here, to the letter (except changing the username, on that tutorial the username created to run OctoPrint is "octoprint." Really doesn't matter what the username is as long as you remember to change the username where applicable.

Once you get everything installed per the tutorial correctly, you should be able to run octoprint manually with no issues (from a SSH terminal) 

Note: "pi" is the username OctoPrint is installed to, probably "octoprint" if you follow the tutorial exactly.

`$ /home/pi/OctoPrint/./run`

If that works and starts octoprint, the do the following below.

Make the startup script
I copied this from a tutorial to auto run VNC on startup, and just kept srewing around with it until I got it to work. If this is something dangerously stupid, or wrong... sorry... I did the best I could by googling. This is what happens when you just google stuff until it works. For all i know this is a huge security flaw. I don't know, use at your own risk...
Here's where I copied it from:
http://raspberrypi.stackexchange.com/questions/23134/how-can-i-get-tightvnc-to-start-automatically-on-startup

Open nano

Paste this into it

```
#!/bin/sh
### BEGIN INIT INFO
# Provides:          octoprint
# Required-Start:    $local_fs
# Required-Stop:     $local_fs
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Start/stop tightvncserver
### END INIT INFO

# Set the USER variable to the name of the user to start tightvncserver under

export USER='pi'

eval cd ~$USER

case "$1" in
  start)
   sudo su - $USER -c "/home/pi/OctoPrint/./run" 
    echo "Starting OctoPrint server for $USER "
    ;;
esac
exit 0
```

Ctrl X to exit and save it as /etc/init.d/octoprint

`$ chmod 755 /etc/init.d/octoprint`

`$ update-rc.d octoprint defaults`

If that doesn't work:

`$ update-rc.d /etc/init.d/octoprint defaults`

That didn't work for me it was doubling the directory.

It now starts on boot every time. Thank you everyone for your help.

guysoft, 2017-02-18:
@poorredneck The post above kinda defies the idea of OctoPi. Why give people complex instructions?

poorredneck, 2017-02-18:
I don't understand what the issue is. I have found many other posts on various forums where people cannot get autostart to work on a manual installation of Armbian and I feel pretty f***ING proud of myself for figuring it out, considering I am a newb to linux. I simply posted what I did to get it to work, in the hopes some other noob like me will find it on Google and get it to work.

If it's complex please offer a better alternative, that is new user friendly (IE step by step)

Many, many tutorials for things Linux related are assuming people who are reading them are all ready experts on the command line or Linux in general. This is why many people stick with Windows and Mac. It's too intimidating to even get started, then you get frustrated and quit.

If I misinterpreted your post I apologize.

guysoft, 2017-02-18:
@poorredneck 
Here you go, this is a build:
https://storage.googleapis.com/octoprint/Armbian_5.25_Orangepilite_Ubuntu_xenial_default_3.4.113_desktop-0.14.0.zip

md5:
https://storage.googleapis.com/octoprint/Armbian_5.25_Orangepilite_Ubuntu_xenial_default_3.4.113_desktop-0.14.0.zip.md5

I'm sorry I don't have all the time to explain. I am just also have a startup company to run, and this is all volunteer work, I not getting paid for this. I relay on other people making guides, explaining etc. Either that or I have no time to maintain OctoPi.
I can tell you there is some code I need to commit to get this to work, there is a bug in pip/configuration thing which I need to report (will link here when I do). Its not for the newbies (yet). But its cool you are figuring this out on your own. I know there is a learning curve.

guysoft, 2017-02-18:
This is the bug in pip: https://github.com/pypa/pip/issues/4290

anthonywob, 2017-03-11:
@guysoft @universam1 
I'm trying to work through this Octopi Install. I flashed the image guysoft provided in the comments for the Orange Pi Pc. It booted as usual change passwords etc (Using a monitor). I've tried to locate the device on my network but it doesn't show up. octopi.local doesnt work either. 
So how do I access octoprint? I'm assuming that once I can connect via network simply going to the IP address in my browser will pull up Octoprint and I can begin the setup as usual. Is this correct?
If so I'l keep troubleshooting my network connection - any tips in that department will also be appreciated.
Thanks.

guysoft, 2017-03-11:
@anthonywob You need to set up your network connection, if you are having issues setting up wireless I suggest you try first an ethernet connection.
Also what is the output of ``sudo ifconfig -a``?

anthonywob, 2017-03-11:
@guysoft I'm not using wifi. Currently have Ethernet connected. I attached an image of the output as I can't connect via SSH yet as it's not on the network. 
[image]

guysoft, 2017-03-11:
Perhaps your network is on ipv6? 

What is the output of ``sudo mii-tool eth0``?

anthonywob, 2017-03-11:
Im not sure. 
Output to that command is:
negotiated 100baseTx-FD, link ok

anthonywob, 2017-03-11:
I checked my router and I have not enable ipV6 so I should still be on ipV4. 
test-ipv6.com confirms I'm still on ipV4 - if this is what you mean.

anthonywob, 2017-03-11:
Also, looks like the orange pi is showing up in my LAN connected devices
with only this:
FE80::D84D:2AFF:FEAF:9F8	unknown	DA:4D:2A:AF:09:F8	Ethernet	0-00-00 00:00:00.00
but not IP address usually there are two entries:
192.16.0.#   "the name of device" "MAC address" "connection type" "expiration"
(then a repeat with the IP address changed to FE80:: (numbers and letters) "device name" etc

anthonywob, 2017-03-12:
So it seems for some reason my router isn't assigning this device an ipV4 address only an ipV6 even though it assign ipV4 for all my other devices.
Using the ipV6 i was able to SSH into the orangepi and set it up.
using:
ifconfig eth0 192.168.0.100
I can manually assign an IP address and connect through my browser but this last until a reboot, on rebooting the ipV4 is lost.
Is there a way to permanently assign the ipV4 address?
Tomorrow I will work on getting a webcam linked - if the one I have lying around ends up working.

guysoft, 2017-03-14:
This seems to be worth to [ask on Arabian fourm](https://forum.armbian.com/index.php?/forum/13-allwinner-h2h3/) why that happens. Its going a little beyond the knowledge I have of their system. It might be something they came across.
You can set up a static IP using ``/etc/network/interfaces``. [Here are instructions on that](https://www.swiftstack.com/docs/install/configure_networking.html#changing-network-configuration).

gyrex, 2017-06-07:
Hi, I'm trying to build a version of octopi from Armbian_5.27_Orangepilite_Debian_jessie_default_3.4.113 and I'm getting the following error message:

```
'/home/john/git/OctoPi/src/variants/armbian/filesystem/root/etc/wpa_supplicant/wpa_supplicant.conf' -> './filesystem/root/etc/wpa_supplicant/wpa_supplicant.conf'
cp: missing destination file operand after 'usr/bin'
```

Any ideas?

guysoft, 2017-06-07:
@gyrex I think its faling on the line 
```bash
#black magic of qemu-arm-static
cp `which qemu-arm-static` usr/bin
```

which means you don't have ``qemu-arm-static`` installed, which is a requirement listed in the README

gyrex, 2017-06-07:
@guysoft Thanks for your prompt response! I'm trying to build this in a vm using an i386 version of ubuntu and can't find qemu-arm* in apt-get. Will this work or do I need to build this on an ARM system?

guysoft, 2017-06-07:
Hey, 
The nightly builds are also done on a i386 machine. I suggest you [use the vagrant build method as described here](https://github.com/guysoft/OctoPi#building-using-vagrant)

gyrex, 2017-06-07:
Thanks for your help @guysoft OK, the package on ubuntu is "qemu-user-static" and I got a little further but now I'm getting another build error. Build log attached.

[build.log2.txt](https://github.com/guysoft/OctoPi/files/1058472/build.log2.txt)

Any ideas mate?

guysoft, 2017-06-07:
@gyrex For some reason, that image does not have i2c installed, and therefore no i2c user group. 
So you will need to [remove the ``,i2c`` from the line in this file](https://github.com/guysoft/OctoPi/blob/devel/src/variants/armbian/pre_chroot_script#L11)

gyrex, 2017-06-07:
@guysoft Do you mean my ubuntu system doesn't have an i2c user group or the armbian image? I'll remove the line in any case and let you know how it goes.

guysoft, 2017-06-07:
No, that user is in the armbian image.
Please use the vagrant build method to remove all build environment
difference.

gyrex, 2017-06-07:
I can't run vagrant because I require a licence for a vmware provider apparently. I removed i2c from pre_chroot_script and I'm still getting error messages. Build log attached.

I think I'll install a fresh version of ubuntu armbian on my OrangePi and try building it on that.

[build.log3.txt](https://github.com/guysoft/OctoPi/files/1058695/build.log3.txt)

gyrex, 2017-06-07:
@guysoft I tried building this on the OrangePi using the Ubuntu xenial build and it failed. build log attached.
[build.log4.txt](https://github.com/guysoft/OctoPi/files/1058802/build.log4.txt)

guysoft, 2017-06-07:
1. Use virtualbox provider,  its free software
2. Use the Jessie image

gyrex, 2017-06-07:
1. I'm not running this in a virtual environment. It's running natively on the OrangePi Lite running Armbian_5.27_Orangepilite_Ubuntu_xenial_default_3.4.113_desktop
2. I'm trying to build octopi using the "Armbian_5.27_Orangepilite_Debian_jessie_default_3.4.113" image

I don't know why it's not building :(

gyrex, 2017-06-08:
@guysoft have you got any other suggestions mate? Sorry to bug you but I'm not well versed on build environments. I really appreciate your help :)

guysoft, 2017-06-08:
Are you building this on the orangepi or on a different machine?  It should be built on a different machine.
Please use vagrant

hogtave, 2017-06-10:
Hi everyone,
Thanks guysoft and everyone helping linux newbs like me!
Is there any chance someone built Octopi for the Orange Pi zero using either the Debian or Ubuntu [image](https://dl.armbian.com/orangepizero/)?
I tried both build method without any success. Building directly on the Opi0 gives me an error  ("losetup: /dev/loop0: detach failed: Device or resource busy"), so i tried the vagran method (using a VM) but couldn't make it work ("Couldn't open file /home/vm/Octopi-armbian/src/base" when sudo vagrant up).

[4 more replies]

closed 2018-01-04 by guysoft