guysoft/OctoPi #233: Octopi don´t start with plugged Camera

MaxMihawk, 2016-06-11:
Hey guys. My Pi doesn´t start correctly if my camera is plugged in the USB hub. So i have 2 cases:
**Case 1:** I´ll boot the Pi without cam. Then i connect the webcam via usb. All works like a charm.
**Case 2:** I´ll boot the Pi with the cam already connected. I can´t even log in.

Webcam lsusb --> 0c45:62f1 Microdia (=logilink UA0075)
/boot/octopi.txt --> camera_usb_options="-r 640x480 -f 5 -y" (cpu usage at 25% for mjpeg-streamer)
RasPi3 with Octopi 0.13 (with update, upgrade, fileexpand, thats all)
Powersupply have 5V - 2,1A (needs 5,2V - 0,4A messured)

```
Common error message is: (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
mmc1: Timeout waiting for hardware interrupt.
mmc1: Timeout waiting for hardware interrupt.
```
or:
```
Fixing recursive fault but reboot is needed!
A start job is running for dhcpcd on all interfaces (1...33s / 1min 33s)
```
I can buy another cam, but is my cam able to tell my Pi "DON'T BOOT 3ITCH!" ? Rly? I´m not clear about that. Any tips?Hey guys. My Pi doesn´t start correctly if my camera is plugged in the USB hub. So i have 2 cases:
Case 1:
I´ll boot the Pi without cam. Then i connect the webcam via usb. All works like a charm.
Case 2:
I´ll boot the Pi with the cam already connected. I can´t even log in.

Webcam: 0c45:62f1 Microdia (=logilink UA0075)
/boot/octopi.txt --> camera_usb_options="-r 640x480 -f 5 -y" (cpu usage at 25% for mjpeg-streamer)
RasPi3 with Octopi 0.13 (with update, upgrade, fileexpand, thats all)
Powersupply have 5V - 2,1A (needs 5,2V - 0,4A messured)

```
Common error message is: (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
mmc1: Timeout waiting for hardware interrupt.
mmc1: Timeout waiting for hardware interrupt.
```
or:
```
Fixing recursive fault but reboot is needed!
A start job is running for dhcpcd on all interfaces (1...33s / 1min 33s)
```
I can buy another cam, but is my cam able to tell my Pi "DON'T BOOT 3ITCH!" ? Rly? I´m not clear about that. Any tips?

seefood, 2016-06-11:
a wild guess - it IS a PSU issue. it may only suck 400 mA on idle average, but the startup process is a bit more consuming, and maybe there just isn't enough power for the initial kick at boot.
Solutions: a powered hub, or maybe leaving the camera as one of the 4 directly-connected devices? buying a different camera is probably not going to be a solution...

MaxMihawk, 2016-06-12:
At the startup, my amperemeter said: 0,09A rising to 0,4A during the boot process. Actually i don't have an active usb hub for testing :/ . I tried with another PSU (5,3V 2A) and a cable with 10 cm. In my opinion, the Pi need about 1A? and the USB hub provides a max of 0,5A to the cam. So I´m at 1,5A max (all other units are unplugged, except the display with his own supply). I´ve done a fresh installation (overwriting the SD befor), same problem. Thanks for the tip, i´ll try it with a bigger PSU or an active USB hub if i get one somewhere, but in my opinion this will not solve the problem. (I wish, you are right :) ).

diomark, 2017-01-11:
Did you ever find a solution to this? I have the exact same problem. VERY annoying:(

MaxMihawk, 2017-01-12:
I´ve added a 3A 5V stepdown modul to my PSU. After this, it works for a few weaks in my testsetup. But the last times i test it, i had the same problem again -.-" . Now i´ve ordered a usb hub with an on / off button from china. Still not arrived :) .