guysoft/OctoPi #234: General API key (not user API key) gets reset to N/A

w1ll1am23, 2016-06-13:
I have been running OctoPi for a couple months now. I have another application setup to monitor the printing via the OctoPrint API. Occasionally I will stop receiving updates in the application and when I check in the OctoPrint API settings the API key is set to the string "N/A" I will put the old API key back and things will be normal again for awhile. I have not been able to narrow down when this happens, but it is annoying. I just upgraded to the latest OctoPrint this weekend and the issue remains. I am not sure if this is an OctoPi or OctoPrint issue. If I should open an issue on the OctoPrint page, please let me know. Thanks.

pcbroch, 2016-06-13:
I have been noticing the same problem this weekend.  I thought the upgrade was the culprit but if you had the problem before the latest, I guess it's not the upgrade.

w1ll1am23, 2016-06-13:
I did notice it right after the upgrade, but I also noticed it before that. Maybe it happens on a restart of the server?

pcbroch, 2016-06-13:
Good idea.  I'll mess with it when I get home tonight.

w1ll1am23, 2016-06-24:
Had a chance to look into this? I tried a couple restarts and didn't notice
an issue, must be something else that is causing the problem. Or a
combination of things.

pcbroch, 2016-06-24:
It hasn't happened again for me.  Rebooted a few times as well with no issues...  Odd.

jjkd, 2016-10-07:
Just happened to me, googled and found this issue report.

guysoft, 2016-10-20:
What version of OctoPrint are you using? it might be an OctoPrint issue.
@foosel any bug reports on your end?

sdenovan, 2016-11-13:
I ran into this and was able to get the API key to reappear by clearing the browser cache and restarting (Chrome on Ubuntu 16.04). Restarting Octoprint / the Raspberry Pi didn't do anything.

arjhun, 2017-01-06:
Hello People, 

I found out that the "Last Pass" chrome extension was messing with my input field for the API key. Disabling the extension resolved the problem. Check if you have an extension running that auto fills or alters your input fields in any way. Hope this helps.

sky4055, 2017-01-28:
@arjhun , same issue with PassIFox Extension on Firefox. Disabling the extension resolved the problem.

foosel, 2017-02-28:
Closing since the OctoPi bug tracker is the wrong location for this. There's a ticket about this on the OctoPrint issue tracker now: foosel/OctoPrint#1751

closed 2017-02-28 by foosel