guysoft/OctoPi #238: Raspberry Pi 3 - Wifi Channels 12 13 and 40

MarcusWolschon, 2016-06-28:
With the OctoPi image 0.13.0 as it is,
even after setting the Wifi-country to "Germany" via 
   sudo raspi-config 
and rebooting, Wifi networks on channel 40 are not visible to
   sudo sudo iwlist wlan0 scanning | grep ESSID
and Wifi channels 12 and 13 are not in the list returned by
   sudo iwlist wlan0 channel

MarcusWolschon, 2016-06-28:
`iw list
Wiphy phy0
	max # scan SSIDs: 10
	max scan IEs length: 2048 bytes
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports roaming.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
		 * P2P-device
	Band 1:
		Capabilities: 0x1020
			HT20
			Static SM Power Save
			RX HT20 SGI
			No RX STBC
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * tdls_oper
		 * start_sched_scan
		 * start_p2p_device
		 * crit_protocol_start
		 * crit_protocol_stop
		 * connect
		 * disconnect
	Supported TX frame types:
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * managed: 0x40 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
	valid interface combinations:
		 * #{ IBSS, managed, AP } <= 2, #{ P2P-client, P2P-GO } <= 1, #{ P2P-device } <= 1,
		   total <= 3, #channels <= 2
	Device supports scan flush.`

TuftyB, 2016-06-28:
Can you check the country code with:
iw reg get
Typically should be using the 2-character code for the country you are in (i.e. DE for Germany. You can find the list of ISO 3166-1 alpha-2 codes on the Wikipedia page [ISO 3166-1 alpha-2](https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2).

MarcusWolschon, 2016-06-29:
<pre>
pi@octopi:~ $ iw reg get
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 40), (N/A, 20), (N/A), NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

pi@octopi:~ $ cat /etc/default/crda 
# Set REGDOMAIN to a ISO/IEC 3166-1 alpha2 country code so that iw(8) may set
# the initial regulatory domain setting for IEEE 802.11 devices which operate
# on this system.
#
# Governments assert the right to regulate usage of radio spectrum within
# their respective territories so make sure you select a ISO/IEC 3166-1 alpha2
# country code suitable for your location or you may infringe on local
# legislature. See `/usr/share/zoneinfo/zone.tab' for a table of timezone
# descriptions containing ISO/IEC 3166-1 alpha2 country codes.

REGDOMAIN=
</pre>

That is after setting the Wifi-country in rasp-setup and rebooting.

TuftyB, 2016-06-29:
My initial thoughts are that this isn’t specifically an OctoPi issue/fault, it appears to be a unique client specific case.
My technical expertise is in networking and as someone who hasn’t used OctoPi/OctoPrint/Raspberry Pi/Linux, my advice is limited in this area.

To assist, I see a couple of concerns:
  1) Why are you using ch40, this is in the 5Ghz range which does not appear to be supported by your wifi device (only 2.4Ghz channels are listed in quoted ‘the iw list’ output)?
  2) What are you trying to achieve by changing the regulatory domain?

Please note:
Linux allows changing regulatory domains in compliance with worldwide regulatory restrictions and to achieve this, devices always respect their programmed regulatory domain and a country code selection will only enhance regulatory restrictions. Therefore if your device was programmed for operation in the US which allows operation on channels 1-11 on the 2.4 GHz band and you visit Japan which allows operation on channels 1-14 and you change your regulatory domain to JP you will not be able to use channel 12, 13 or 14, however if you have a device programmed for operation in Japan and visit the US and you select US as your regulatory domain, then channels 12-14 will be disabled.

Therefore you need to check the firmware coded region of your WiFi device to determine it’s reg-domain.

Also as your device is set to country "00", which is the "world regulatory domain" and contains generic settings. If you ned to set this and are using CRDA, then it may be misconfigured and I would recommend that you look at:
https://wireless.wiki.kernel.org/en/developers/regulatory/crda

Hopefully there are a couple of pointers in above that might direct you to solution.

MarcusWolschon, 2016-06-30:
a) OctoPi is an entire Image containing Octoprint. The image is outdated. That is an issue that needs to be fixed for users to have a great experience using it. (setting the regulatory domain to Germany worked after hacking temporary internet access via Ethernet and thus updating outdated packages of the OctoPi image.)
b) It's not my private Wifi, so I have no choice in how it is configured and it is my only means of providing permanent Internet access for the OctoPi.
c)  By changing the regulatory domain to my country I *did* achieve that it allowed the Wifi driver to use channels such as 40 that are legal in my country and actually used for the Wifi here but forbidden by default because they are not legal in ALL countries of this planet.

guysoft, 2016-06-30:
@MarcusWolschon 
a)
The latest OctoPi iamge is about a month old. There is only one raspbian image ahead of it, and its release notes does not say anything about wifi changes. So unless you can confirm that works Its likely not the case

MarcusWolschon, 2016-06-30:
What I did:

* apt-get update
* apt-get dist-upgrade
* wait for 2 dozen packages to update
* reboot
* sudo raspi-config to set Wifi-County
* reboot
* enjoy working Wifi

Maybe the Raspbian image is not up to date with the Raspian packages? I'd guess this issue would be Raspberry Pi 3 specific and many people are not affected or would not go to the length of filing a change request to see it fixed in the image after they finally found out the cause of their wifi issues.

foosel, 2016-06-30:
The raspbian image most likely is not up to date with the raspbian package repository since keeping packages up to date is what apt get is for. It is unfeasible putting out a new image every day just in case that someone somewhere fixed an issue in one of the hundreds of packages included in a default raspbian install. That goes for both stock raspbian as well as OctoPi. Package managers fulfill this specific purpose of keeping stuff up to date incrementally.

I'd actually suggest addressing this specific problem with the raspbian people though because they are the ones who can actually fix this hen-egg-problem in their base image (and document it properly in the release notes, something I also still have a sneaking suspicion of not being reliable). It is not possible for octopi to become a full fledged standalone Linux distribution - we simply lack the resources for that (@guysoft has a company to run and my time is already taken up fully with OctoPrint itself) and have to depend on a stable base. So please help raising awareness at the origin of that stable base about issues with it instead of ripping our heads off.

rurwin, 2016-10-26:
There were exactly these problems on the Raspberry Pi kernel in the March-April timeframe. Those took a while to get through validation and into the released Raspbian image. So a month-old image in June may still have been broken. I would expect any images now to be fixed.

See: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=44044&start=475

The Pi 3 doesn't do 5GHz; the hardware doesn't support it.

guysoft, 2018-01-27:
Is this fixed then?

guysoft, 2019-07-02:
Closing due to no reply, new Pis have 5GHz support.

closed 2019-07-02 by guysoft