guysoft/OctoPi #241: OctoPi doesn't reconnect to wifi once the connection is lost

marian42, 2016-07-03:
Wifi connection is established at boot. When restarting the router, the wifi host is disabled for a few minutes. Once the wifi is up again, the Pi doesn't connect. This makes OctoPi impossible to use and the only fix is rebooting, which cancels the current print.

Salandora, 2016-07-04:
Which version of OctoPi do you have?

guysoft, 2016-07-06:
Closing if there is no reply.

marian42, 2016-07-06:
The last release seems to be March 21. I downloaded OctoPi after that.

guysoft, 2016-07-06:
SSH to your pi and type: ``cat /etc/octopi_version``
Alternatively paste the name of the file you flashed. I have on grantee you indeed downloaded 0.13.0.

@Salandora spesifically wants to know because there was a bug fixed in the last version which was released AFTER March 2, the date is of the last git commit, not the release date.

Also, there are other basic things to rule out - What RaspberryPi version are you using, and how much current is your power supply?

marian42, 2016-07-07:
I had an outdated version.

closed 2016-07-07 by marian42