guysoft/OctoPi #243: Having problem in connecting octopi from internet, Help

joinalahmed, 2016-07-06:


joinalahmed, 2016-07-06:
@foosel

foosel, 2016-07-06:
Nobody can help you if you do not describe your problems *fully*, and tagging people won't help that either. Not being able to connect to your OctoPi instance from the internet also is a support question, *not a bug*. Ask in the [G+ community](https://plus.google.com/communities/102771308349328485741) or on the [mailing list](https://groups.google.com/group/octoprint) for this. A bug tracker (like this one) is not a support forum, and there will be more people available to help in the actual support channels (the linked G+ community and mailing list). But again, just stating "It doesn't work, please help" won't allow anyone to actually help you, you'll need to clarify what you tried, how you tried it, provide your configuration attempts etc etc.

closed 2016-07-06 by foosel

joinalahmed, 2016-07-06:
i'm able to access octoprint in my local wlan easily,  but having problems in accessing fron outside, i'm able to reach upto my router using dynamic dns but not able to reach my octopi

guysoft, 2016-07-06:
You need to set up a DMZ, this is done in your router and therefore its an issue unrelated to OctoPi, you can see talks about it at #225.