guysoft/OctoPi #245: Camera: am I missing something obvious?

davidgewirtz, 2016-07-08:
I have a couple of spare webcams around, one is a no-name and one is an IPEVO. I know neither are among the wiki listed, but I found someone online who posted somewhere (can't find it again, natch) who said he got it working fine with OctoPi.

When plugged in with the no-name (640x480 craptaculous no-name webcame) in a USB port, OctoPi woulnd't boot. When plugged in with the IPEVO, it's booting, but I'm not seeing an image.

In any case, using the latest OctoPi release, is there any specific setup required other than just plugging the camera in and rebooting? Is there any diagnostic process to go through to tell if the system is seeing the camera. Using Raspberry Pi 3 with the CannaKit 2.5A supply.

Thanks!
--David

guysoft, 2016-07-09:
Can you provide the output of ``lsusb`` with the camera plugged in?
And also ``lsusb -n`` ?

davidgewirtz, 2016-07-09:
lsusb results in:

Bus 001 Device 005: ID 27b1:0001
Bus 001 Device 004: ID 1778:0204
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -n isn't supported. I've got version lsusb (usbutils) 007. Options are:

Usage: lsusb [options]...
List USB devices
  -v, --verbose
      Increase verbosity (show descriptors)
  -s [[bus]:][devnum]
      Show only devices with specified device and/or
      bus numbers (in decimal)
  -d vendor:[product]
      Show only devices with the specified vendor and
      product ID numbers (in hexadecimal)
  -D device
      Selects which device lsusb will examine
  -t, --tree
      Dump the physical USB device hierarchy as a tree
  -V, --version
      Show version of program
  -h, --help
      Show usage and help

Thanks!

--David

P.S. The only other USB device plugged in is the LulzBot Mini.

davidgewirtz, 2016-07-11:
Here's a -v dump, if that helps any:

Bus 001 Device 004: ID 27b1:0001
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x27b1
  idProduct          0x0001
  bcdDevice            0.01
  iManufacturer           1
  iProduct                2
  iSerial               220
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           62
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0
      CDC Header:
        bcdCDC               10.01
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
[... 180 lines cut ...]
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

davidgewirtz, 2016-07-11:
Okay, some more detective work. The IPEVO P2V is device 004, and it shows in the summary, but errors in the verbose listing. According to http://www.linux-hardware-guide.com/2014-02-02-ipevo-point-2-view-p2v-usb-document-camera, the IPEVO should be supported by uvc, but I'm not finding it in dmesg at all.

So, any thoughts? Do I need to download or install some other packages?

Thanks!

--David

davidgewirtz, 2016-07-15:
For those curious, I asked the folks at IPEVO, and this was their reply (and it worked):

Please change the mjpg_streamer parameter in OctoPi from -f 10 to -f 30. You can change this by going to /root/bin, edit webcamd, modify the value of camera_usb_options, and change the -f 10 to -f 30. After that, you will need to restart the device or reactivate webcamd service in order for it to work.

foosel, 2016-07-15:
Please do *not* edit webcamd. There is a config file for changing things like frame rates, resolution etc at /boot/octopi.txt (or alternatively directly in the root of the card when you use it as a thumb drive, same thing as with the network config). See also https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-change-the-webcam-resolution-on-octopi

davidgewirtz, 2016-07-16:
Thanks, Gina. I made the change (removed the incorrect one and then
followed your instructions). Also, you might find this interesting. I
published it on ZDNet this morning:

http://www.zdnet.com/article/how-to-add-a-smartphone-controlled-brain-to-your-3d-printer-for-about-fifty-bucks/

--David

*David Gewirtz*
Educator, Advisor, Columnist, Geek
Read my columns at DIY-IT <http://www.zdnet.com/blog/diy-it/> and ZDNet
Government <http://www.zdnet.com/blog/government/>
Read my lab notes at ZATZLabs <http://zatzlabs.com/lab-notes/>

Office: +1 202-642-5627 / Mobile: +1 321-722-4620

<https://mail.google.com/http://twitter.com/davidgewirtz>
<https://mail.google.com/https://www.facebook.com/davidgewirtz>
<https://mail.google.com/https://www.linkedin.com/in/davidgewirtz>
<https://mail.google.com/http://youtube.com/davidgewirtztv>

foosel, 2016-07-18:
@davidgewirtz, awesome, and thanks for the nice shoutout! :) 

Regarding this issue, is there anything left unsolved now or can the ticket be closed? In general you probably have higher chances of getting help with stubborn hardware on the [mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741) btw, since this bug tracker here is mostly read by a handful of development interested people only, whereas the other two locations also have a lots of users reading and replying. Larger group = higher chance of someone having encountered and solved the same hardware issue :)

Btw, when you reply to the mails from Github, your replies get automatically added to the ticket as well. I took the liberty to remove your email footer from your reply, since it contained some contact information I was not sure you actually wanted out in the open.

davidgewirtz, 2016-07-18:
Thanks, Gina. Much appreciated (the contact cleansing) and all your work. The tweaks for the camera work. It's actually quite nice because it's got a stand meant for document imaging so it makes locating the camera quite easy. Plus, it was in my closet!

And yes, I consider this a closed issue.

--David

closed 2016-07-18 by davidgewirtz