guysoft/OctoPi #246: OctoPi Webcam Unuseable

Hotwire, 2016-07-10:
I have an OctoPi server setup on a Pi 2. I have a Logitech C510 webcam setup with it, and it recognizes the camera, and displays images... images is all. It seems that the video has a refresh rate of once ever 2 seconds. its set to the standard 640x480 and 10 fps. It should be running MUCH smoother then that should it not?

guysoft, 2016-07-10:
1. Are you sure your wifi is sufficient and not too slow? (Aka everything
else is fast).
2. Can you run top and see if rhe CPU is below 100%

markwal, 2016-07-10:
Perhaps, depends on what format the camera produces and how much processing is required to produce the mjpeg stream from the format produced by the camera.  However, my guess in your case is that you're saturating the available bandwidth of the WiFi dongle given its connectivity with your AP (position, walls, RF noise etc.)

You can test my theory by removing the dongle and plugging in Ethernet directly to your router reboot and test using new ip address.

Hotwire, 2016-07-10:
top - 20:39:48 up 19 min,  1 user,  load average: 0.20, 0.16, 0.23
Tasks:  86 total,   1 running,  85 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.1 us,  0.2 sy,  0.0 ni, 99.6 id,  0.0 wa,  0.0 hi,  0.2 si,  0.0 st
KiB Mem:    883056 total,   155872 used,   727184 free,    13732 buffers
KiB Swap:   102396 total,        0 used,   102396 free.    62608 cached Mem

Everything else is super fast, I can stream video from my media center to ANY tv in the house. But I cant get this webcam to do much more then stills every 2 seconds

markwal, 2016-07-10:
Yeah, but do any of them have as bad of an attenae as your wifi dongle as your pi?  And keep in mind, the mjpeg stream is "optimized" to be as fast as possible to compute and is also a fairly fat in terms of bytes. There is *no* interframe compression.

Hotwire, 2016-07-10:
The ChromeCast probably has one just as bad. And it streams everything just fine. I have 4 of them in various parts of the house. much further away and more walls and floors in the way. Im kind of at a loss with it. I can get pretty smooth video when I have it set to 80 x 40 resoluion.

jerm, 2016-08-08:
What would lend to a bandwidth issues. I have a similar problem in the workspace of my apartment.

Try this: From a command prompt on the Pi, run
   sudo apt-get install speedtest-cli
   speedtest-cli

What do the results say?

totalitarian, 2017-01-11:
I have the same issue on my RPi2. The speed test gives me this

```
Testing download speed................................................................................
Download: 23.27 Mbit/s
Testing upload speed....................................................................................................
Upload: 5.05 Mbit/s
```

jerm, 2017-01-11:
what do you get on your normal machine?

totalitarian, 2017-01-11:
About the same. Not quite sure how this helps as I normally stream over internal wifi, not via my ISP

jerm, 2017-01-11:
Understood.  That's just a quick and easy test. If it were vastly different that's be an easy warning sign.  

You can check machine<->machine throughput with iperf for nuttcp (http://nuttcp.net/nuttcp/nuttcp-8.1.4/binaries/).

daktylus, 2017-11-12:
Yes that is the sad truth. mjpg-streams are fat. there is hardly any compression. I currently run 640x480 @ 10 fps and it needs about 4.5 Mbit/s . Unusable if I want to access it via the web.

I don't understand the issues, but it seems complicated to implement the usage od the h.264 compression on the pi. (With that, 720p @30 fps shouldn't be way less than 4 MBit/s.)

guysoft, 2017-11-12:
I've been using remotely the telegram plugin, ``/status`` gives you a nice picture of the printer.
Closing if no reply from original posters

daktylus, 2017-11-13:
Is there a chance that a better video compression will be implemented?? 

I found this comprehensive discussion: https://raspberrypi.stackexchange.com/questions/7446/how-can-i-stream-h-264-video-from-the-raspberry-pi-camera-module-via-a-web-serve    Maybe the WebRTC is worth trying.

guysoft, 2019-04-23:
@daktylus makes this related to:
https://github.com/guysoft/OctoPi/issues/529

and:

https://github.com/guysoft/OctoPi/issues/545

filipgoc, 2019-05-12:
AstroBox (a heavily divergent octoprint fork) uses webrtc if anyone felt like exploring that route. 
https://github.com/AstroPrint/AstroBox

guysoft, 2019-05-12:
@filipgoc Good to know - any idea where the sources for their video side? I see only the octoprint side.

@foosel Do you know where to poke to see what they are using to stream? 
``octoprint/timelapse.py`` seems to still have your name on top and ffmpeg.

foosel, 2019-05-13:
I have no idea. They forked years ago, no idea what they did in the meantime since contrary to made promises I never got any contributions back.

guysoft, 2019-05-13:
@CoDanny Hey, any input by any chance on how Astroprint does camera streaming using WebRTC?

BTW, just pulled the latest image and ran it using CustomPiOS ``qemu-boot.sh``. The issue.txt file says:
```
pi@astrobox-2741:~$ cat /boot/issue.txt 
Raspberry Pi reference 2017-11-29
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b71bf3179310aadf9aaf2ffcdf094a88e488dce8, stage2
```
Which is kinda strange can't tell how the image is built because there is not source available, that would suggest though someone is manually upgrading an exiting image.

filipgoc, 2019-05-13:
I just noticed that it has a WebRTC message when loading/ failing to load a video stream, and when I searched the repo there were some mentions, so I thought it reasonable to think it does something with it.