guysoft/OctoPi #249: OctoPi, OctoPiPanel, and Raspberry Pi 3?

CJeffyB, 2016-07-20:
I have a Raspberry Pi 3, and have been using it with OctoPi happily for several months.  I was recently gifted a PiTFT, and have been completely unable to make it work with OctoPi.  Something seems to be hanging on boot, but I have no idea what I need to do next to attempt to track down the problem.

First, I started with my existing install, and ran the ./enable-adafruit-pitft script.  Rebooted when prompted, and the device promptly hung.  The TFT shows a solid grey-white screen.

I then pulled the Micro SD card, and did a completely fresh install.  Ran the script again, and it is hanging in the exact same place.

The system does not respond to anything on the web interface, nor to SSH requests.

Is there something that I'm missing or need to do to make it all work?

Are there any steps that I can take to provide further information to assist troubleshooting?

guysoft, 2016-07-21:
Suggest you post to the OctoPiPanel issue tracker:
https://github.com/jonaslorander/OctoPiPanel/issues

CJeffyB, 2016-07-21:
Thanks!  I wasn't sure if it was a problem specific to OctoPi, or if it were an OctoPiPanel issue.

I've posted it here:
https://github.com/jonaslorander/OctoPiPanel/issues/51

closed 2016-07-21 by guysoft