guysoft/OctoPi #252: Trying to rename serial ports so multiple machines can be plugged in and selected independently

timmit99, 2016-08-02:
I've been trying to use my OctoPi instance on a few printers (never more than one at a time) but when i go to select a serial port all i see is /dev/ttyACM0 , /dev/ttyACM1, ...etc.

I started down the path of giving each USB device a static name folowing solutions from  [Stack Exchange](http://unix.stackexchange.com/questions/66901/how-to-bind-usb-device-under-a-static-name) and [here](http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/). My rule in /etc/udev/rules.d/99-usb-serial.rules is 
`SUBSYSTEM=="tty", ATTRS{idVendor}=="2a03", ATTRS{idProduct}=="0042", SYMLINK+="RampsBoard"`

it works on the RasPi, entering `ls -l /dev/RampsBoard` returns `lrwxrwxrwx 1 root root 7 Aug  2 17:40 /dev/RampsBoard -> ttyACM0` but only ttyACM0 shows up in the octoprint interface.

markwal, 2016-08-02:
You can list additional ports for the drop down in settings.  OctoPrint looks for the typical port names by wildcard which doesn't match your custom name. (BTW this is an OctoPrint thing, not OctoPi which is the image that includes OctoPrint and other stuff like mjpg-streamer)

JustAnother1, 2016-08-03:
/dev/serial/by-id/ ?

timmit99, 2016-08-03:
Ok i got it working by adding /dev/wilsonII to the additional serial ports. I posted it here because i wasn't sure where my issue was, whether it was in the octoprint software or if it was something else i needed to do relating to the image and Raspberry Pi.

Thanks!

closed 2016-08-03 by timmit99