guysoft/OctoPi #257: Webcam not working

hackermagnet, 2016-08-18:
I have installed octopi on a raspberry pi 3. A couple months ago I had installed it on a raspberry pi 2 (can't remember which version) and my webcam was working fine. Now it doesn't work (it's the same webcam) and dmesg gives me this:
```
[10293.856434] usb 1-1.5: new high-speed USB device number 4 using dwc_otg
[10293.989950] usb 1-1.5: New USB device found, idVendor=18ec, idProduct=3299
[10293.989973] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10293.989986] usb 1-1.5: Product: USB2.0 PC CAMERA
[10293.989998] usb 1-1.5: Manufacturer: ARKMICRO
[10294.023253] media: Linux media interface: v0.10
[10294.048668] Linux video capture interface: v2.00
[10294.155339] uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (18ec:3299)
[10294.155511] usbcore: registered new interface driver snd-usb-audio
[10294.161462] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[10294.164307] input: USB2.0 PC CAMERA as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input0
[10294.164415] usbcore: registered new interface driver uvcvideo
[10294.164420] USB Video Class driver (1.1.1)
```
Is there any fix?

guysoft, 2016-08-18:
What version are you using now on the pi2?

hackermagnet, 2016-08-18:
I don't use pi2 anymore. I have the 0.13.0 on pi3

guysoft, 2017-07-23:
Any news on this? If not, closing due to inactivity and no camera version listed.

closed 2019-07-02 by guysoft