guysoft/OctoPi #259: upgrading to  newest version is not available

sputtering, 2016-09-04:
I am running 1.2.13 and I get a notification that 1.2.15 is available but no checkmark (not able to update directly). I tried running this from SSH but it errored out:

```
git clone https://github.com/guysoft/OctoPi.git
cd OctoPi/src/image
curl -J -O -L  http://downloads.raspberrypi.org/raspbian_latest
cd ..
sudo modprobe loop
sudo bash -x ./build
```

Not sure if this is an issue with OctoPi or OctoPrint but I wanted to get the ball rolling and find where the issue is.

markwal, 2016-09-05:
If all you want to do is update OctoPrint to the latest, the ssh steps would be:

ssh pi@octopi.local (or whatever your octopi address is)
cd OctoPrint
git pull
source ~/oprint/bin/activate
python setup.py clean
python setup.py install
sudo service octoprint restart

foosel, 2016-09-05:
> I get a notification that 1.2.15 is available but no checkmark

Make sure the update configuration is alright then, because it sounds like it isn't. Settings > Software Update > wrench icon in the upper right corner, should say ``/home/pi/OctoPrint" for "OctoPrint checkout folder" in the dialog that opens.

It gets shipped that way on OctoPi, but maybe it got changed somehow.

guysoft, 2016-09-05:
Are you trying to build OctoPi or upgrade OctoPrint from an OctoPi
installation?

sputtering, 2016-09-05:
@foosel That is the path that is there. It is also set to Release for the Version Tracking.

@guysoft Since I was having issues trying to upgrade the normal way, I was trying to upgrade through SSH.

Cyberman-tM, 2017-01-17:
I think I have a similar issue.
OctoPrint version is apparently 1.2.10 , a fresh install of "octopi-jessie-lite-0.13.0" . I get the message that 1.3.0 (stable) is available, but soon a fail when trying to update.

Trying to parse through the log, I found this:
[code]
ValueError: ("Expected ',' or end-of-list in", 'OctoPrint>=1.2.6+', 'at', '+')
[/code]
Is 1.2.10 simply too old to be updated to 1.3?

What's weird is that I decided to try the update again, but deleted the log file first (via UI) - apparently now the log file isn't generated anymore?

foosel, 2017-02-28:
Anyone running into issues with updating Octo*Print* via the built-in Software Update Plugin, please provide a full [bug report](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) on OctoPrint's bug tracker.

This is the issue tracker of Octo*Pi*, the SD image, which comes preconfigured in a way that updates should work out of the box just fine. If they don't, either something in the configuration got messed up (hence check that as described above) or you are running into some bug or other kind of issue in Octo*Print*. In either case, the OctoPrint bug tracker is the better suited location (but DO provide all required information and stand by for further follow up questions).

closed 2017-02-28 by foosel