guysoft/OctoPi #260: Pine A64

zachk56, 2016-09-14:
Hello everyone, I was wondering if I could get some clarification on if the Pine A64 Single board computer that was recently made available this year, could run OctoPi.

I tried following the video guide on how to do it and followed in exactly, but when I plug my Pine in nothing happens at all. I looked up another guide which made me install Debian, then made me use the terminal to get OctoPi from the repository. It worked for about 2 minutes then I couldn't access it. If I could get any ideas I'd truly appreciate it!

[Here](http://forum.pine64.org/showthread.php?tid=2064) is the guide that worked briefly

seefood, 2016-09-14:
I have a Pine A64+ at home I can test with, but isn't that a bit of an overkill for OctoPrint? I got it for more demanding projects, that seems like a waste...
Anyway, I don't know if it will work with the OctoPi 32bit images. better install OctoPrint on a made-for-pine image of Debian or Ubuntu.

zachk56, 2016-09-14:
Well, I originally did not intend to get the Pine A64+ 2gb for this, but I really am sick of dedicating my laptop to prints.. I only have a laptop and the Pine. Thank you for your help and clarification. If you do plan on testing it, please let me know! I just want to make sure it's me and not the guide! (Probably is me though!)

guysoft, 2016-09-14:
AFAIK the stock OctoPi should not work.
However, there is a variant that runs armbian for OctoPi. And Pine 64 is supported in Armbian. If you build the Armbian variant with the Pine 64 image it should work. See #232

guysoft, 2018-01-04:
Closing because armbian works

closed 2018-01-04 by guysoft