guysoft/OctoPi #261: Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.

knifefish, 2016-09-27:
What were you doing?

Updated to newest version of octopi on raspberry pi 3. Can no longer connect to printer.

What did you expect to happen?

Everything to work.

What happened instead?

Cannot connect to printer. Prusa i3 mk2

Branch & Commit or Version of OctoPrint

1.2.16

Printer model & used firmware incl. version

Prusa I3 Mk2 - Fimware 3.0.8

Browser and Version of Browser, Operating System running Browser

Chrome - 53.0.2785.116 m (64-bit) - Windows 10 64bit

Link to octoprint.log

 https://gist.github.com/knifefish/b79477672db39e0455ec93c347a71148

Link to contents of terminal tab or serial.log

 https://gist.github.com/knifefish/936be79e30e86baed74413e2e66d3a92

guysoft, 2016-09-27:
Are you using the nightly builds or 0.13.0?

foosel, 2016-09-27:
That has nothing to do with OctoPi itself, and from the looks of it also not with OctoPrint but simply with the printer not replying after a wrong baudrate was selected, which makes auto detection fail.

There's already a ticket for that on the OctoPrint issue tracker which I've also replied to (see foosel/OctoPrint#1517), closing this one.

closed 2016-09-27 by foosel

guysoft, 2016-09-27:
There might be a driver change if he us using the nightly

foosel, 2016-09-27:
log file doesn't look like a driver change but like a picture book case of "wrong baudrate selected"