guysoft/OctoPi #263: build get stuck during git cloning

make-ing, 2016-09-30:
Hei,

i have problems building a standard OctoPi image. It's getting stuck during git cloning commands. Its not always the same repo but always at some clone command. I was not able to successfully build one image. I'm using the newest jessie image from 23.09.16.

guysoft, 2016-09-30:
That is an issue I had too, the way I found around it is to use the vagrant build option. 
Its one of the reasons I built it. [Instructions how to use it are here int he README](https://github.com/guysoft/OctoPi#building-using-vagrant)

make-ing, 2016-10-05:
Hi guysoft,

i tried to get it to work with vagrant (1.8.6) the hole day, but without success. I will open another issue for this.

guysoft, 2017-07-23:
Closing due to inactivity, the vagrant build works for the past year, also had other people report success.
If you want you can re-open and I can continue to help you, we progressed the past year.

closed 2017-07-23 by guysoft