guysoft/OctoPi #264: vagrant provisioning not working

make-ing, 2016-10-05:
I tried to build with vagrant, but during provisioning vagrant stopped with an error:
```
Vagrant was unable to mount VirtualBox shared folders. This is usually
because the filesystem "vboxsf" is not available. This filesystem is
made available via the VirtualBox Guest Additions and kernel module.
Please verify that these guest additions are properly installed in the
guest. This is not a bug in Vagrant and is usually caused by a faulty
Vagrant box. For context, the command attempted was:

mount -t vboxsf -o uid=1000,gid=1000 vagrant /vagrant
```
I assume this is a bad image without guest additions installed. Does someone have a similar problem?

Im using vagrant 1.8.6 with virtualbox 5.0.24.

make-ing, 2016-10-05:
It seems to work with the box: wzurowski/vivid64

I will test it further and make a pull request later.

guysoft, 2016-10-05:
Great, thanks for sharing all this!
It works for me, but ill have to check what visrtualbox version I have.
Also the image might have changed.
Open a pull request and I'll see if your solution work here too.

guysoft, 2016-11-21:
Fixed, working here, merged, all is good

closed 2016-11-21 by guysoft