guysoft/OctoPi #266: Unable to connect Wifi via octopi-network.txt

niksoley, 2016-10-06:
I configured the octopi-netwok.txt and still without wifi, I tried different configurations with and without security. Reading the forum, I've seen people having problem with the mac texteditor, so, after coping the image, I also tried editing through terminal SSH, with double and single quotes.
Searching the internet, I found this tutorial that worked for me (https://www.youtube.com/watch?v=dZeJb6Jv-9E), he had similar issue, and was able to solve it by modifying the interfaces and wpa_supplicant.conf files. Although it worked for me, it doesn't look the right way of doing it, also, the tutorial shows how to do it for WPA-PSK, and in my case, I want to a open wireless network.
Follow my specs:
Mac OS Sierra, 2016-03-18-octopi-jessie-lite-0.13.0.img, Raspberry Pi 2 Model B, Wifi dongle Vilros 150 mbps (it came in a official rasp kit, seems to the Realtek RTL8188CUS-GR model), Router TL-WR940N / TL-WR941ND.
Tks in advance.

niksoley, 2016-10-12:
I just edited thew pa_supplicant.conf file with:
network={
	ssid="plaintext-test"
	key_mgmt=NONE
}
and its working fine. However, I could not make this work through octopi-netwok.txt.

ekaggrat, 2016-11-11:
i am having the same issue. I tried two different cards and still wont work. I uncommented the wired connection and hooked it up to my laptop and got it to work.. Then i edited the wpa_supplicant.conf file and added the wifi configuration. but on next reboot the rpi refuses even to boot . The only way to get it to work then is to wipe out the entire card and redo everything ... 

any clues what am i doing wrong?

Salandora, 2016-11-11:
@ekaggrat with out your modified interfaces and wpa_supplicant.conf file it's hard to guess, (of course wipe out all the personal informations like key and ssid)

It could also be that the Pi is taking longer to boot because it tries to establish a connection but fails for some reason so it would try until a specified timeout ~60s.

Atleast we need more informations to help you here.

chaseacton, 2016-12-09:
Having the same issue. It can scan for networks fine, but doesn't connect to the network specified in the supplicant file.

jimmcmillan, 2016-12-20:
I had the same issue following these instructions (https://learn.adafruit.com/octoprint-on-m3d/octoprint-set-up) using textedit on macOS Sierra. I followed the same instructions and edited the file using "Textmate" and wifi worked fine.

Xserxes, 2017-01-05:
Same issue on my side, configured the txt file on boot partition, but not connecting. After some editing i got it working with the wpa_supplicant.conf.
Also, sometimes wireless is lost, if i use my webcam.Strange
I am using an Raspberry 3b with WLAN onboard.

NCBob, 2017-01-15:
I've had the same issue, installing a standard Jessie image wifi worked fine, wouldn't work with Octopi with either the built in or an adapter.  I Followed the youtube instructions to disable the octopi-network.txt source load and updated my supplicant config manually and it's working fine now.

rordaz, 2017-01-22:
Had the same problem, but reading this post gave me some clues where the problem was...

Do a sudo nano /boot/octopi-network.txt and you'll see the double quotes in your wpa-ssid, wpa-psk are wrong, copy them from the lines bellow and it will work!. Apparently text editor uses the wrong type of double quotes, maybe due to text encoding done by TextEdit.app on the mac.

if you don't know how copy and past in nano, here are the keyboard shortcuts
https://wiki.gentoo.org/wiki/Nano#Cut.2C_copy.2C_and_paste

NCBob, 2017-01-22:
I had already tried editing it with nano, gedit and vi, none of them worked

jedi5200, 2017-02-18:
I tried everything in the books, but something simple helped me. like rordaz said, the thing the did it for me was deleting the mac-made quotations and replacing them with pi-made quotations, one set each around the ssid and psk.

rordaz, 2017-02-18:
Glad to help @jedi5200! - What printer are you using? just for curiosity

guysoft, 2017-02-18:
@LeonFedotov Might work for you too

ARKopp, 2017-02-19:
editing the quotations on the pi also worked for me. Come on can somebody but this on the octoprint website? I am trying since hours to get this working.

NCBob, 2017-02-19:
It's not part of octoprint, it's part of octopi

John-Mc, 2017-04-01:
FYI - you can "fix" TextEdit on the Mac so it doesn't use those "curly" quotes:
Launch TextEdit, go to TextEdit>Preferences. In the box that pops up, select "Plain Text" and deselect "Use Smart Quotes" and deselect "Use Smart Dashes". You should be able to update files without TextEdit making those annoying substitutions.

NCBob, 2017-04-02:
Yeah, but I'm not using text edit, I'm using visual studio code and making sure it's utf-8 format

mhermanns, 2017-04-25:
I tried all the above, but nothing worked. First I thought, that I fixed the problem, when I so the mangled double quotes Mac Textedit left in the octopi image. But then I figured, that the Pi couldn't see my SSID - my AP was set to channel 13 (EU) and the image can only see channels 1 to 11:
`sudo iwlist wlan0 channel`
Switched my wifi channel to 1 and the Pi connected instantly.

jameswendt, 2017-07-21:
I fixed the quotes per the suggestion above and it worked! Thank you!

ofloveandhate, 2017-09-01:
@NCBob is [this](https://www.youtube.com/watch?v=dZeJb6Jv-9E&t=37s) the video you used?

ofloveandhate, 2017-09-01:
is the fundamental problem here that the quotes **as distributed** in octopi-network.txt are not actually quotes, but so-called smartquotes, and they must be manually replaced?

John-Mc, 2017-09-02:
No, the quotes as distributed in the image are correct. Apple's TextEdit program (and perhaps some others?) change them into smart quotes if you edit any text adjacent to them. By default, if you try to type quotes in, TextEdit will use smart quotes (unless you've disabled that "feature" in TextEdit as described above).

thomllama, 2017-09-23:
just a quick note/info...
If using mac, just download "text wrangler" god, trying to think how long i've been using it and ca't remember, but pretty much since the beginning of OSX-ish time frame..  it works perfectly, Use it for editing firmware files also (marlin) instead of through Adreino as you can have old/new open and copy things over.. Oh, it's free and in the App store. :)

robineter5, 2017-12-12:
@mhermanns Your method worked for me, my router channel was set to 13 while the raspberry supports only channels 1 to 11. I guess you are a hero now. Thankyou.

drcameron, 2017-12-13:
Solved my issue using the information here: https://github.com/RetroPie/RetroPie-Setup/wiki/Wifi#manual-configuration-interfaces. I was using WEP and so the quotes are NOT needed. In fact with them I could not connect. Without them, I connected instantly. 
The octopi-network.txt file needs to read as below for WEP. I imagine WPA should work for most as noted above.

iface wlan0 inet manual
   wireless-essid SSID
   wireless-key PASSWORD

sodle, 2017-12-30:
Additional pitfall: the Pi 3's built-in WiFi only supports 2.4GHz networks. Sorry if it's off topic, but I came across this thread while troubleshooting my setup and that turned out to be my issue.

ckopotic, 2018-03-08:
This took me a while to figure out. To set up wifi:
After flashing the SD card with octopi open it on your computer before inserting it into the PI. Find the "octopi-wpa-supplicant.txt" file
Open it in notepad or similar and replace it with (you may need to change the country and key_mgmt depending on your network):

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
 ssid="NAME"
 psk="PASSWORD"
 key_mgmt=WPA-PSK
}﻿

Foxies-CSTL, 2018-04-20:
Hi,
My building block for WPA2 with FR channel (13) only into /boot/octopi-wpa-supplicant.txt,not change into /boot/octopi-network.txt ;-)
**network={
   ssid="xxxxxxx"
   psk="pass"
   proto=RSN
   key_mgmt=WPA-PSK
   pairwise=CCMP TKIP
   auth_alg=OPEN
 }**

It's fine for me

calston, 2018-05-01:
For some reason the broadcom firmware for the Pi 3 is missing on the current image... Pro stuff

guysoft, 2018-05-01:
@calston Please use 0.15.0. We removed ``octopi-network.txt`` Use now ``octopi-wpa-supplicant.txt``

Also closing this issue.

closed 2018-05-01 by guysoft

OutsourcedGuru, 2018-12-28:
@guysoft If the `octopi-network.txt` file is still part of the image, you may want to delete it to avoid confusion.

guysoft, 2018-12-29:
We left it with instructions to use the other file. We could decide to
remove it cc @foosel

OutsourcedGuru, 2018-12-29:
Noting that if you google "octopi-network.txt" you might end up on an old Adafruit tutorial.

John-Mc, 2018-12-29:
I think it's good as-is at this point, since a number of the help videos still send people to octopi-network.txt.  Better to have them open a file that directs them elsewhere than have them not find the file at all.

foosel, 2019-01-10:
> I think it's good as-is at this point, since a number of the guides and help videos still send people to octopi-network.txt. Better to have them open a file that directs them elsewhere than have them not find the file at all.

That was my reasoning behind *not* removing it but leaving it with instructions to use the new file. There are indeed a ton of (often wrong) guides out there that still refer to this file, and removing it I'm sure will lead to people just manually creating it and then wondering why nothing works and opening tickets here.

calston, 2019-01-10:
> That was my reasoning behind _not_ removing it but leaving it with instructions to use the new file. There are indeed a ton of (often wrong) guides out there that still refer to this file, and [...]

Maybe there needs to be a better official setup doc for it. Or maybe it's time to build some system configuration UI stuff into Octopi so people can just set it up there. In fact, you could even default the Octopi to run an access point with DHCP so people can connect directly to it for initial config like other IoT devices.

calston, 2019-01-10:
I have time, I'll see if there's a more appropriate ticket to discuss further

foosel, 2019-01-10:
> In fact, you could even default the Octopi to run an access point with DHCP so people can connect directly to it for initial config like other IoT devices.

On Pi3 (& the not recommended 0W) you can. On earlier models you can't. Keep that in mind. And yes, there *are* users installing on hardware like that. So there should still be a configuration fallback.

calston, 2019-01-10:
> On Pi3 (& the not recommended 0W) you can. On earlier models you can't. Keep that in mind. And yes, there _are_ users installing on hardware like that. So there should still be a configuration [...]

Earlier models did not have wifi, but yes you absolutely can detect a USB wifi dongle and use that instead.

guysoft, 2019-01-10:
As I said - there is work needed to be done and I have less time for it.
PRs are welcome, and I already wrote a module you can start with.

calston, 2019-01-10:
> Famous last words ;) I went down that road around 2014 already. It nearly cost me my sanity. Not detecting the WiFi dongle, no... detecting whether the chipset supports ap mode to begin with and if so what hostapd binary or driver to use for it. The issues that caused was basically why I gave up on netconnectd, as promising as it was when you were running supported hardware.

Okay sure, it is a bit pot-luck. But I don't think it's enough of a deal breaker to forgo the option entirely

guysoft, 2019-01-10:
I think its acceptable only supporting newer models, the main issue is that
if wifi fails you loose connection to your printer OctoPi puts up a wifi
hotsopt. So you need to check periodically if internet is back and
reconnect.

I am willing to put the module in an experimental release if someone wants
to work on it.

John-Mc, 2019-01-10:
I used a hotspot following setup instructions written by @kennethjiang that worked well for me on older OctoPi versions. Unfortunately, it does not run under Raspian Stretch. I also only ever tried it on an RPi 3, which was the most recent model at the time. It might be a great option to help with initial set-up, but the only problem I've ever had with that was trying to use Apple's TextEdit without making the appropriate changes to keep it from corrupting the quotation marks.

I'd love to see a hotspot feature now, but my needs were for other than initial set up (helping out in a school that had issues with sharing their WiFi password, even though this was for a classroom activity supported by a very tech-savvy teacher).

OutsourcedGuru, 2019-01-10:
> Maybe there needs to be a better official setup doc for it. Or maybe it's time to build some system configuration UI stuff into Octopi so people can just set it up there. In fact, you could even default the 

I recently created [this](https://github.com/OutsourcedGuru/octopi-install), if it's helpful.

OutsourcedGuru, 2019-02-13:
[Another one confused](https://community.octoprint.org/t/octopi-wifi-network-connection-troubleshooting-megatopic/2608/182?u=outsourcedguru) by octopi-network.txt.

guysoft, 2019-02-14:
You could move it in to the wiki here if that helps more people find it, but not sure how to improve communication, as the one seeing all the issues I always get the feeling people don't read, but thats because of where I stand.