guysoft/OctoPi #267: Support latest Chromium browser

BillyBlaze, 2016-10-12:
The latest chromium-browser has been added to the main repository of Raspbian Jessie. However due to the outdated kernel this package cannot run on a OctoPi image. Running this will produce something like:

````
/usr/lib/chromium-browser/chromium-browser: symbol lookup error: /usr/lib/chromium-browser/chromium-browser: undefined symbol: mmal_vc_init_fd
````

The current solution is to update the kernel by downloading ```rpi-update``` and running that. I have instructed my Wiki with these steps, but it would be awesome if it works out-of-the-box.

Thanks.

guysoft, 2016-10-20:
New nightly uses latest base kernel, and should work now.

closed 2016-10-20 by guysoft

BillyBlaze, 2016-10-21:
Thanks you sir 🙇

tamboer, 2017-08-01:
Wow. Struggled with this for a while. Thanks it worked.

anugrahasinha, 2019-04-15:
This is a very old thread, but I stumbled of the problem today. `rpi-update` solved the problem,