guysoft/OctoPi #268: gcode Sent to OctoPi Now Fails

ManeBjorn, 2016-10-21:
I have been using OctoPrint with my QIDI Dual Extruder printer for a few months with no issues.
Now about 2 weeks or so ago any gcode files I send to be printed fail. They pause after a few mins and fail.
Even clicking restart on the front of the printer it will heat back up and fail.

I use Simply 3D and never had an issue. I can take the x3g file saved at the same time as the gcode file put on an SD card and it prints fine.
I've made a new image of OctoPi and still have the issue. I even uninstalled S3D and retried. Same issue.
I've even tried with the MakerBot software.
Just can't print with OctoPrint anymore.
Even gcode files I have sent prior and printed with no issues fail as well. 
Anyone have any suggestions?
Thank you.

guysoft, 2016-10-22:
Not sure, sound like a OctoPrint issue. 
How much space is left on on the SD card on the raspberrypi?

ManeBjorn, 2016-10-23:
It has 26+ gb left on it. I actually went and created a new one on a new card and RPi3 just to be sure it was not hardware. Same issue. It is so bizarre.
I will ping the OctoPrint git. I thought it was possibly and issue with the OctoPi part of it but as you said might not be.
Thank you for responding.

kretu, 2017-05-15:
I've got similar problem. If i upload gcode after reboot first print is ok but i can't reprint it. I must delete old and reupload the same gcode to be able to print. And some time even that isn't enough and i have to reboot printer and reupload again. Fortunately it shows error of wrong buildsize when selecting model so i don't start the print. (If i forgot about that ekstruder works strange and gets cluged every time at beginning of a print) (CTC bizer printer)

markwal, 2017-05-19:
This definitely isn't an issue with OctoPi.  Plus, I think there are two separate issues here.  One is the issue about build size which apparently is caused by a G92 in the start gcode being interpreted (foosel/OctoPrint#1906). 

The other isn't explained very well here, but I've seen reported by a CTC user on the forums and is likely an interaction of the GPX plugin and your start gcode.  If you do decide to report it on the markwal/OctoPrint-GPX bug tracker, please include your start gcode and describe specifically in what way the print fails: doesn't start, prints in the wrong position, terminal window shows all the commands going by, but the printer isn't moving, etc.  @guysoft I'd recommend closing this one since unlikely to be anything to do with building the OctoPi image.

closed 2017-05-19 by guysoft