guysoft/OctoPi #269: [request] Auto WiFi Hotspot if no ssid found on rp3

polyarmid, 2016-10-25:
something like this :

http://www.raspberryconnect.com/network/item/315-rpi3-auto-wifi-hotspot-if-no-internet

guysoft, 2016-10-30:
There are several attempts to do this, @foosel had one while ago here is the [newDevel branch](https://github.com/foosel/OctoPi/commits/newdevel) commit she had:
https://github.com/foosel/OctoPi/commit/50fef2fd7b8529717ecdd78816d26282f8fca80b
And also recently @make-ing [committed something to a netconnectd branch](https://github.com/make-ing/OctoPi/tree/netconnectd).

No one fully implemented it yet.
If anyone wants to put the time in it, it would be super cool and we could make it usable in other projects.

John-Mc, 2017-04-01:
Kenneth Jiang is working on this as part of his Painless WiFi for OctoPrint:
[https://medium.com/@kennethjiang/painless-wi-fi-for-octoprint-4e6b68005400](url)
Scroll down to near the bottom. It still has a couple of bugs he is working out, but I think he's close. (He is starting from a more recent version of OctoPi than the 0.13 release. I don't know if this is the 0.14rc1 release or not).

guysoft, 2017-04-02:
@foosel That looks pretty easy to make/make a variant for Pi3/Zero W.
Think we should do that for 0.15.0?

John-Mc, 2017-04-02:
@foosel    Kenneth just started working on this a week or so ago at my request. I installed his "full-featured slicer" plugin, had a few questions, which turned into a hour+ long phone conversation about how we were using OctoPrint in the elementary school where I'm volunteering (helping with their 3D printer and their makerspace). He asked what kinds of plugins we would like to see in the future, and our discussion wandered away from plugins into other features that would help in our setting. 

Since we may want to move the printer around to different schools and networks, and our school district's IT guy has not been cooperative, I asked if it were possible to implement an "AutoHotspot if no known WiFi networks are found" in OctoPi. I pointed him to one that someone had made that worked in a regular installation of Raspian-Jessie, but with my limited knowledge I could not get to work on OctoPi/Jessie-Lite.

He started working on it. My role has been the Noob who is sometimes very good at breaking things, then he tells we what commands to enter so I can send him info to find out why it broke.

Hochachtungsvoll
John-Mc

John-Mc, 2017-04-02:
FYI - Kenneth Jiang has updated the info on his page about the AutoHotspot linked above. The only change was to the /usr/bin/autohotspot part. He has a good degree of confidence in this now.