guysoft/OctoPi #270: serial communication problems after Update / or with nightly builds

DomLuu, 2016-10-27:
Hi 

I use octopi-jessie-lite-0.13.0.zip on my PI 3 the last weeks. I have some WLAN problems (lost connection after a day or two), so i did a apt-get update/upgrade/dist-upgrade as recommended in some sites if you have WLAN problems.

After this I got serial communication timeouts while printing. The printjob starts, pause after 5 sec. and start again a few sec. and so on.  
I tried the nightly build and have the same problems without forcing an update.

So I go back to the jessie-lite-0.13 - it runs perfekt but this horrible WLAN Problems .... hmmm

regards Dominik

guysoft, 2016-10-27:
Have you tried setting the timout communication time in OctoPrint?

DomLuu, 2016-10-27:
Yes - still communication problems, but only with the 0.14 nightly build or with 0.13 and forced update via shell command apt-get update/upgrade/dist-upgrade

guysoft, 2016-10-27:
Have you tried setting the communication timeout in OctoPrint.

DomLuu, 2016-10-27:
to which value? the defaults works perfekt - just after the update I got this problems on serial port.

guysoft, 2016-10-30:
Just double them for a start. On my printer I had to make them larger

foosel, 2016-10-30:
If the default values work with the printer in question on 0.13 but not 0.14 (or 0.13 after a dist upgrade), it's not the printer or OctoPrint or the timeouts, it's something caused by the update.

guysoft, 2017-07-23:
Any news with this? If not, closing due to inactivity