guysoft/OctoPi #271: [1.3.0rc1] Websocket Issues via Internet

FinstererFred, 2016-11-02:
Hi, i installed 1.3.0rc1 yesterday and it works great via wlan, 
but when i try to connect over the internet, the page loads and i get this error in the javascript console
`VM54:35 WebSocket connection to 'ws://xxx.de/sockjs/122/fo05oigz/websocket' failed: Error during WebSocket handshake: 'Sec-WebSocket-Accept' header is missing`, 
and the "Server seems to be offline" popup.

This happends in latest Chrome and Firefox. Strangely it seems to work with Chrome from my 
Android cellphone.

foosel, 2016-11-02:
Wrong issue tracker ;) I suggest you get in touch on the OctoPrint mailing list or in the G+ community (both are linked from the website, I'm on mobile right now so I can't provide the links here) since this sounds more like a configuration issue with your forwarding chain or maybe even some new browser requirements that will necessitate changes in forwarding instruction, rather than a bug in either OctoPrint or OctoPi. When you do, please provide full information on how exactly you have set up your forwarding chain, including any configuration files from reverse proxies (haproxy, nginx, ...) that might be in the mix and also the port forwarding rules you have setup. Also include how you are calling up OctoPrint's interface (https vs http, any sub paths...). Without that information it will be impossible to analyse this issue.

This bug tracker here is for the sd card image you installed which among other things contains OctoPrint. Your issue seems to be with OctoPrint, or rather with web socket confections from outside your LAN, not the sd card image per se.

FinstererFred, 2016-11-02:
Ah crap, sorry. I just googled octoprint bug report and followed the wrong link.

Thanks for the hint, i will post this in the right bugtracker and provide more info. :)

closed 2016-11-02 by FinstererFred