guysoft/OctoPi #273: Serial Port Disappears

aggiehavoc, 2016-11-03:
Hi all,

I have been using OctoPrint/OctoPi for several weeks now.  I am using a Raspberry Pi B Rev. 2 with a Maker Select V2.  I have previously been able to connect to the printer using /dev/ttyUSB0, but that is suddenly unavailable on the dashboard screen.  The Pi sees the printer for sure (lsusb), and I am able to connect to and control the printer using Repetier-Host from there, so it's not the printer.  Further, in the settings area of OctoPrint, the printer is in fact available to select, but selecting it there seems to have no impact on the main dashboard page availability.  Any help would be greatly appreciated.  Thanks!

Please let me know if I need to provide any further details.

aggiehavoc, 2016-11-03:
Oh, and I'm on OctoPrint version 1.2.16.  I have the latest release of OctoPi.

guysoft, 2016-11-03:
Does the file ``/dev/ttyUSB0`` exist when OctoPrint does not see it?

aggiehavoc, 2016-11-03:
Yes, it does.

guysoft, 2016-11-03:
Then its not really an OctoPi issue, its an OctoPrint issue and should go
in https://github.com/foosel/OctoPrint/issues/

aggiehavoc, 2016-11-03:
Further info: I did find it via ls command, but also in the settings area, if I unplug /dev/ttyUSB0, it is removed from the list.  It's just the dashboard that doesn't show it.

closed 2016-11-03 by aggiehavoc