guysoft/OctoPi #274: Network setup wifi/cable combined?

Helmi, 2016-11-07:
Hi,

trying around with the octopi-network.txt file for quite a while now. My target is to get a static IP on wifi and one on cable while cable shall only be used when plugged in. I got it working now with a static Wifi IP but only if the static section is commented out. I've set wlan0-octopi to static instead of manual and added address, netmask, broadcast and gateway. This works but the pi still seems to get a DHCP address additionally.

Is there any documentation I missed about setting a static IP for wifi or am I just the only one with such a weird setup? :)

Thanks!

guysoft, 2016-11-07:
Regarding static IP, there is this issue:
https://github.com/guysoft/OctoPi/issues/202

As far as I know, if the ethernet device is plugged in and gets a DHCP connection then it will be the main connection, but I am not 100% sure.

guysoft, 2018-04-10:
Closing if no reply

closed 2022-01-09 by guysoft