guysoft/OctoPi #275: can't access at octopi.local or ip address

oakfielduk, 2016-11-14:
Hi,

I have just acquired a shapeoko cnc machine with 3d printer attachment.
I have being having a lot of trouble trying to get things working....

I have the Raspberry Pi B loaded with Octopi,
I have it connected to wifi through an apple extreme router (also tried it through iPhone personal hotspot)
I have found the IP address
I can ping the IP from my laptop (running windows 7 and Internet Explorer, also tried Firefox)
I have TightVNCViewer installed and it connects to the Pi just fine.
I just seem to fall at the last hurdle - when I type in octopi.local to the address bar I get nothing and it's the same when I type 10.0.1.23 - the ip address assigned to the Raspberry Pi.

Please can someone help - it feels like it should be a very simple solution but I have wasted far too much time on this now.

Thanks,
Mark.

guysoft, 2016-11-16:
What version of OctoPi are you using?

ARKopp, 2017-02-19:
I have the same issue. It was with the news 0.13 version

guysoft, 2017-02-19:
... What version of OctoPi are you using?

ARKopp, 2017-02-19:
the newest on the website 0.13

guysoft, 2017-02-20:
Can you ssh to it?
@ARKopp I gather you dont have the tightvnc configuration set like the original poster? Can you please give more details?

oakfielduk, 2017-02-21:
I'm sorry, I completely forgot I started this post. 
After wasting a couple of days trying to get this to work, I resorted to using wires from my laptop using octoprint. 

If I can give any details that might help to diagnose the problem then please let me know.

ARKopp, 2017-02-21:
I got it to work by putting in the  ssid and password on my pi. I then ssh like this ssh pi@octopi

guysoft, 2017-02-21:
@ARKopp what does octoprint.log say then?

casche, 2017-03-11:
@guysoft I have the same problem.

Last 4 lines of log:

```
2017-03-11 17:46:36,008 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-03-11 17:46:36,189 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2017-03-11 17:46:36,244 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2017-03-11 17:46:36,256 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2017-03-11 17:46:36,262 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
```

Version: 2016-03-18-octopi-jessie-lite-0.13.0.img 
Raspberry-pi: Model B powered with a 2.1a 5v psu

I can ssh to the Pi using octopi.local but cannot reach it via the browser.

guysoft, 2017-03-11:
@casche What happends when you run on the raspberrypi (on ssh):
``wget localhost``?

@foosel Is there a way to check locally on the Pi if the service is working?

casche, 2017-03-11:
```bash
pi@octopi:~ $ wget localhost
converted 'http://localhost' (ANSI_X3.4-1968) -> 'http://localhost' (UTF-8)
--2017-03-11 18:58:03--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... failed: Connection refused.
Connecting to localhost (localhost)|127.0.0.1|:80... connected.
HTTP request sent, awaiting response... 
```

guysoft, 2017-03-11:
Can you do ``cat index.html``? So we can see what is inside?

nathan-long, 2017-03-12:
Same issue here with 0.13  Adding a blank file with no extension named "ssh" to the root of the SD card fixed the issue for me.

casche, 2017-03-12:
@guysoft I have no idea what happened but I left it alone for about 5 hours and now it is working.

foosel, 2017-03-13:
> @foosel Is there a way to check locally on the Pi if the service is working?

yes, and @casche already did it just above your post, `ps -ef | grep -i octoprint`. I'd also suggest to check if haproxy is running:

```
pi@octopi:~ $ ps -ef | grep -i haproxy
root       731     1  0 Mar10 ?        00:00:00 /usr/sbin/haproxy-systemd-wrapper -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid
haproxy    732   731  0 Mar10 ?        00:00:00 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -Ds
haproxy    735   732  0 Mar10 ?        00:00:16 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -Ds
pi        2589  2539  0 08:37 pts/0    00:00:00 grep --color=auto -i haproxy
```

and if it's bound to its ports:

```
pi@octopi:~ $ sudo netstat -tulpen | grep haproxy
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      0          11528       735/haproxy
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      0          11529       735/haproxy
udp        0      0 0.0.0.0:50301           0.0.0.0:*                           0          11530       732/haproxy
```

because if you are not getting anything in your browser, it can't be OctoPrint's fault - if it's only OctoPrint that is down, haproxy will produce a nice error page that explains what to do, if you don't even get that, it's haproxy or something else above it that's causing you issues.

> Adding a blank file with no extension named "ssh" to the root of the SD card fixed the issue for me.

That sounds more like you couldn't reach your OctoPi instance *via SSH* (instead of via your browser) because the daemon was disabled - which shouldn't be the case on 0.13 because Raspbian introduced that "let's disable SSH by default and cripple headless setups instead of forcing users to change their passwords" non-sense only after that image got pushed out. However, you might have introduced this behaviour by performing an `apt-get upgrade`, I can't remember right now if that does also disable SSH out of the box. Current nightlies should already ship with a blank `ssh` file on `/boot`/the root of the FAT partition, since OctoPi is supposed to run headless and disabling SSH is outright damaging otherwise due to that.

----

**In general, when you can't reach OctoPrint running on OctoPi in a browser**, try to

  * ping the Pi by its ip address and `octopi.local` - if that doesn't even work it's an issue with your network. Make sure you entered the SSID and password of your WiFi correctly (if you used Textedit on a Mac, it might have mangled the `"` into some fancy UTF-8 quotes instead - fix them and tell your Textedit to behave:
    ```
    # If you use Textedit to edit this file make sure to use "plain text format" 
    # and "disable smart quotes" in "Textedit > Preferences", otherwise Textedit 
    # will use none-compatible characters and your network configuration won't
    # work!
    ```
    You might also want to test a wired network connection to the Pi, to make sure it's not just your WiFi acting up.
  * ping the Pi by `octopi.local` - if that doesn't work it's an issue with your computer not being able to resolve bonjour domains. If you are running Windows you might have forgotten to install the drivers needed for that, there's a link with more info on that in the "How to use it?" section on the OctoPrint download page, the OctoPi download page and the OctoPi README on this very repo. On some routers bonjour name resolution doesn't work at all, in that case you'll need to use the IP or assign a name to your Pi in your router - you'll need to check your router's manual on how to do that.
  * ssh into the Pi by either ip or `octopi.local` - if that doesn't work your Pi has gotten an IP but for whatever reason the SSH daemon seems to not be running. This might happen if you are using a nightly from any time between @guysoft switching to the november raspbian image until maybe two weeks ago, or possibly also if you updated a 0.13 image via `apt-get upgrade`. Power down your Pi, pull out the SD, put it in a card reader into your computer, place an empty file called `ssh` onto the SD, right next to `octopi-network.txt`. Make sure it is called *exactly* `ssh` - not `SSH` or `sSH` or `ssh.txt` or something, but *exactly* `ssh`. Current nightlies of OctoPi should already ship like that.
  * via SSH, run `ps -ef | grep -i haproxy` - if it doesn't show something like the example output in this comment a bit up, for some reason the reverse proxy in front of OctoPrint is not running. Try `sudo service haproxy restart`. Take a look into the proxy's log via `cat /var/log/haproxy.log`, see if there are any errors reported here (when everything is fine, this file should usually be empty, so if it's not empty, that is already a big clue).
  * via SSH, run `ps -ef | grep octoprint` - if it doesn't show a line starting with `/home/pi/oprint/bin/python`, your OctoPrint server is not running. We need to check why. First take a look at `cat ~pi/.octoprint/logs/octoprint.log`, if that contains any errors. Also check if `sudo service octoprint restart` maybe fires it up. If not, stop the service via `sudo service octoprint stop` and start it manually from your SSH console: `~pi/oprint/bin/octoprint serve`. It will probably die with an error, report back with the error.

dobodu, 2017-07-13:
Hi oakfielduk 

I think you just miss the Bonjour Service in order to enable Windows to reach octopi.local.
Let's google a little and try to find the Apple 3.3 Bonjour Service Setup.
This protocole is included in most Linux version while it's not a standard package for Win 7/8/10

Hope it helps

MichaelNewham, 2017-12-01:
Couldn't access octopi.local through browser myself..
I can ping IP AND 'octopi.local', no problem..
Ssh'ing via IP AND 'octopi.local', no problem..
Ran the above commands and **seems like lots of files / folders 'missing'.**
Image i'm using is latest nightly build 2017-11-28_2017-09-07-octopi-stretch-lite-0.15.zip found at http://docstech.net/OctoPiMirror/nightly/

Am gonna try a stable build.

Heres some text from the terminal -

```
pi@octopi:~ $ ps -ef | grep -i octoprint
pi        1827  1810  0 12:33 pts/0    00:00:00 grep --color=auto -i octoprint
pi@octopi:~ $ wget localhost
--2017-12-01 12:33:49--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... failed: Connection refused.
Connecting to localhost (localhost)|127.0.0.1|:80... failed: Connection refused.
pi@octopi:~ $ cat index.html
cat: index.html: No such file or directory
pi@octopi:~ $ ps -ef | grep -i haproxy
pi        1831  1810  0 12:35 pts/0    00:00:00 grep --color=auto -i haproxy
sudo service haproxy restart
[sudo] password for pi: 
Failed to restart haproxy.service: Unit haproxy.service not found.
pi@octopi:~ $ cat /var/log/haproxy.log
cat: /var/log/haproxy.log: No such file or directory

```

GreyArea1966, 2019-04-08:
Try this...worked perfectly for me...

https://github.com/OutsourcedGuru/octoprint-name-resolution-hacks

guysoft, 2019-04-23:
Closing this. This whole issue has drastically improved and was documented the past 3 years. Thanks to @GreyArea1966 pointing out @OutsourcedGuru 's work

closed 2019-04-23 by guysoft