guysoft/OctoPi #276: Attempting to build new image on Rasbian failed

jecevans, 2016-11-19:
While trying to build a new Octopi image on Rasbian I get the following ouput.

Collecting pyasn1
  Downloading pyasn1-0.1.9-py2.py3-none-any.whl
Requirement already satisfied: PyOpenSSL in ./oprint/lib/python2.7/site-packages (from ndg-httpsclient)
Exception:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
    requirement_set.prepare_files(finder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 666, in _prepare_file
    check_dist_requires_python(dist)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/packaging.py", line 48, in check_dist_requires_python
    feed_parser.feed(metadata)
  File "/usr/lib/python2.7/email/feedparser.py", line 177, in feed
    self._input.push(data)
  File "/usr/lib/python2.7/email/feedparser.py", line 99, in push
    parts = data.splitlines(True)
AttributeError: 'NoneType' object has no attribute 'splitlines'

The version I am building off of is 
Linux Rasbian 4.4.32-v7+ #924 SMP Tue Nov 15 18:11:28 GMT 2016 armv7l GNU/Linux

ScottDWebster, 2016-11-20:
I am also seeing this behavior.  I've tried building on raspbian (2016-09-23-raspian-jessie.zip) and from ubuntu 16.04.1 x86.  In both cases I am trying to build an octopi image based on the full raspbian jessie with pixel so I can also make use of the desktop features on the machine as well.

At various times during the attempted build, I see a message like "sudo: unable to resolve host ubuntu-too" but "ping ubuntu-too" finds the correct ip address.
[octopi-build.log.txt](https://github.com/guysoft/OctoPi/files/601832/octopi-build.log.txt)

guysoft, 2016-11-21:
Confirmed, 
First build this started happening at is ``Build #779 (Nov 3, 2016 2:00:01 AM)``

guysoft, 2016-11-21:
cc @foosel

guysoft, 2016-11-21:
Ran a diff against the build outputs. The only diffrence I can spot is that during pip upgrade:
```
Installing collected packages: pip
  Found existing installation: pip 1.5.6
    Uninstalling pip:
      Successfully uninstalled pip
Compiling /tmp/pip-build-ReWmM6/pip/pip/_vendor/html5lib/treewalkers/etree.py ...
  File "/tmp/pip-build-ReWmM6/pip/pip/_vendor/html5lib/treewalkers/etree.py", line 13
    from pip._vendor. import string_types
                           ^
SyntaxError: invalid syntax
```

foosel, 2016-11-21:
Looks like a bug that only occurs with current pip version 9.x (which gets installed a couple steps prior).

I can reproduce (even on an older raspbian I might add) via:

```
sudo apt-get install python-openssl
virtualenv test
./test/bin/pip install --upgrade pip
for pattern in "pyOpenSSL*"; do cp -v -r --preserve=mode,timestamps /usr/lib/python2.7/dist-packages/$pattern test/lib/python2.7/site-packages/; done
./test/bin/pip install ndg-httpsclient
```

I cannot reproduce any more after downgrading pip to 8.1.2:

```
./test/bin/pip install --upgrade "pip>=8,<9"
./test/bin/pip install ndg-httpsclient
```

Looks like pip 9.x hiccups on the `.egg-info` *file* from the PyOpenSSL debian package, or rather doesn't know what to do with it since it expects newer kinds of metadata. 

That whole mess with copying python ssl stuff back and forth was added in order to get a sane SSL environment in urllib3/requests under the (old) python version shipped with raspbian previously (see [here](https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings) for details). What's the currently available one? If it's >= 2.7.9, it might be possible to get rid of that whole madness which would be good.

guysoft, 2016-11-21:
This is the bug in pip: https://github.com/pypa/pip/issues/4059

foosel, 2016-11-21:
No. That's the issue you saw on your Jenkins, but I didn't see it when testing locally and neither did the log provided by @ScottDWebster contain that issue. 

That's not what's breaking installation of ndg-httpsclient here (pyasn1 installs fine btw, it's indeed ndg-httpsclient in the same pip call as asn1 that's causing the problem).

guysoft, 2016-11-21:
Just ran a build with your fix and it fixes it: http://gnethomelinux.com:8080/job/OctoPi/800/console
@jecevans  and @ScottDWebster , Pull from devel and you can confirm

guysoft, 2016-11-21:
Note: we will need to revert this when 9.0.1 comes out.

foosel, 2016-11-21:
Pinning pip to version 8 is NOT a fix. It's a workaround, and pip 9.0.1 is already out and does also show that behaviour. `pip` will keep on bugging people to upgrade it, and that might then cause issues in the field. 

This needs to be solved either by getting rid of that whole manual SSL setup for Python (which again is not necessary IF the current Raspbian ships with a Python version >= 2.7.9 OR at least allows upgrading to one, someone **please test this**, I can't right now!) or if that's not possible finding a (probably more resource intense) solution to the old pyopenssl debian package causing this issue with current pip versions.

You need to solve the problem, not patch up the symptom.

guysoft, 2016-11-21:
@foosel See the issue i posted from pip, its a vendor inside pip which I don't thing we can really control. One alternative would be to pull from https://github.com/pypa/pip/archive/9.0.1.zip .
Trying that now. I guess I am guilty of wanting to fix this quickly because I have a bit of load at work.

foosel, 2016-11-21:
Again, there are TWO issues at work here, and 9.0.1 is affected for both:

1) That `vendor` thing, that doesn't appear to have any real consequences though, other than ugly log messages, and it only occurs in your Jenkins log, it is not happening in the log provided by @ScottDWebster 
2) The issue quoted in the OP, the Traceback caused during install of pyasn1 and ndg-httpsclient (with the latter being the actual culprit, `pip install pyasn1` runs flawlessly). THAT actually has consequences, making the build fail. 

Both issues are completely unrelated! One (the vendor thing) is an actual pip issue, but rather harmless. The other is a compatibility issue between the version of PyOpenSSL manually installed during an earlier step of the build script and pip versions >= 9.0 (including 9.0.1, which is already released and failing, see my posts above), and what's the actual problem that needs to be solved here.

Since current Python version on Raspbian finally seems to be 2.7.9, try removing [these lines](https://github.com/guysoft/OctoPi/blame/devel/src/chroot_script#L71-L91) from the build script INSTEAD of the downgrade/version pin of pip (so making this only read ``pip install --upgrade pip`` again). Try that, see if the build a) succeeds and b) when actually running the image OctoPrint doesn't drown the log in "insecure platform warning" messages.

guysoft, 2016-11-21:
1. Th full log os not provided. Would be helpful if OP provides it. Not
sure about @ScottDWebster 's log.
2. They must be related in some way. When I change the pip line the pyasn1
stops breaking. And it was consist.

foosel, 2016-11-21:
They are both not present in pip 8.1.2, which is what gets installed when you specify to install a pip version >= 8 and < 9 (your commit comment is also wrong in that regard, you pinned the install to 8.1.2, not 9.0.0 like the comment suggests - 9.0.0 would be ">=8,<=9.0.0"). That's the only way they are related, if you can call that related.

ScottDWebster, 2016-11-21:
By the way, "python -V" on the 2016-09-23 Raspbian Jessie (albeit an dist-upgraded version) says "Python 2.7.9".

Visdoom, 2017-08-07:
Hey guys, 
any news on this?

I got the same error when trying to get 
`pip install imbalanced-learn` to work.

Error log:
```
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python3.4/dist-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/usr/local/lib/python3.4/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/local/lib/python3.4/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/local/lib/python3.4/dist-packages/pip/req/req_set.py", line 666, in _prepare_file
    check_dist_requires_python(dist)
  File "/usr/local/lib/python3.4/dist-packages/pip/utils/packaging.py", line 48, in check_dist_requires_python
    feed_parser.feed(metadata)
  File "/usr/lib/python3.4/email/feedparser.py", line 177, in feed
    self._input.push(data)
  File "/usr/lib/python3.4/email/feedparser.py", line 100, in push
    parts = data.splitlines(True)
AttributeError: 'NoneType' object has no attribute 'splitlines'
```
Downgrading to pip 8.1.2 fixed it

guysoft, 2017-08-07:
There is nothing we can really do here other than downgrading to pip 8.1.2. Someone would need to send a pull request to pip's team.

guysoft, 2019-04-23:
Fixed in https://github.com/guysoft/OctoPi/commit/fb7093bfb950dc890323a4607068e8ceb5fa36cd

commit dc66324d (2016-11-21): Workaround pip 9.0.0 that breaks install of pyasn1 fixes #276
closed 2019-04-23 by guysoft