guysoft/OctoPi #277: octoprint on OctoPi 0.13.0 Lite won't run

MattEttus, 2016-11-20:
I just did a fresh install of octopi-jessie-lite-0.13.0.zip on my Raspi 3.  I made no modifications, and I followed the directions.  

I can ssh into it at octopi.local, but octoprint does not start.  Going to octopi.local:8080 works fine and I get access to my camera.  But http://octopi.local doesn't respond.  nmap shows only ports 22 and 8080 open.

I tried going in by ssh and running octoprint manually, and I get the following:

pi@octopi:~ $ OctoPrint/run 
Traceback (most recent call last):
  File "OctoPrint/run", line 21, in <module>
    import octoprint
  File "/home/pi/OctoPrint/src/octoprint/__init__.py", line 4, in <module>
    from octoprint.server import Server
  File "/home/pi/OctoPrint/src/octoprint/server/__init__.py", line 9, in <module>
    from sockjs.tornado import SockJSRouter
ImportError: No module named sockjs.tornado

guysoft, 2016-11-21:
You are not supposed to run that file directly, OcotoPrint from OctoPi runs from a virtualenv. 
It can be run by reaching: ``~/oprint/bin/octoprint``.
@foosel should he post any log?

foosel, 2016-11-21:
Official support channels for OctoPrint issues are the mailing list or the G+ community. And yes, providing as much information as possible (including any log files and running processes and such) helps enormously in getting help when posting there.

MattEttus, 2016-11-21:
Sorry, I didn't see a link to the mailing list on the web page.  Anyway, I
wiped the SD card and ran it without doing a partition resize in
raspi-config  it worked again.  Then did a partition resize and it dies.
Maybe there is something weird about my SD card.  Anyway, I'm past the
issue for now.

Thanks,
Matt

closed 2017-01-11 by guysoft