guysoft/OctoPi #281: Raspberry Pi pulls down printer temps and triggers MINTEMP error.

old-relic, 2016-12-04:
I have been having problems with MINTEMP errors on my PICA/ArduinoMEGA controller (marlin firmware).
I have discovered that whenever I connect the USB cable from the Raspberry Pi to my printer the thermistor temperatures reported on the ReprapDiscount Smart Controller drop several degrees. The measured temperatures can get below the minimum value for MINTEMP and the error triggers.
I saw that the thermistor ADC inputs on the PICA used Vcc, and I thought maybe Vcc was being changed when I plugged in the USB cable, but when I monitored Vcc it stayed steady at 5.031 volts and clearly was not the cause of the problem.
The problem is still there if the Raspberry Pi is powered down and OctoPi is not running.
Can anybody figure out why this is happening?

guysoft, 2016-12-04:
I'd suggest not connecting the VCC of your printer to the pi in that case. I.e disconnection the pin #1 which gives +5 volts.
Also this isn't really an OctoPi issue, its a power issue with the RaspberryPi.

old-relic, 2016-12-04:
Do you mean the pin that provides power via in the USB cable?

old-relic, 2016-12-04:
Thanks for the tip Guy. That solved my problem.
I isolated the power wiring by slicing open the sheath on the USB cable. Then I pried the braid and foil wrap to expose the red power wire, I cut the red power wire and insulated the ends with a blob of glue and stuffed the wires back into the sheath. 
I couldn't fire a ready solution in the way of a data only cable or adapter.

closed 2016-12-04 by old-relic