guysoft/OctoPi #282: Unable to delete folders

herrerodani, 2016-12-09:
I recently updated to 1.3.0 (master branch). I'm able to create folders but unable to delete them. When I click the trash icon, nothing happens.

herrerodani, 2016-12-09:
Ok. I'm seeing that that problem only happens to folders created with spaces in it's name. For example: I create a folder named "Imperial Assault" then Octopi renames it to "Imperial_Assault" and I'm unable to delete it.

guysoft, 2016-12-09:
cc @foosel

foosel, 2016-12-09:
Just tested (folder name "test 123"). Created, deleted, no issues.

foosel, 2016-12-09:
Also: wrong bug tracker, @guysoft can't do anything about issues within OctoPrint. Please report in OctoPrint's tracker with full ticket, follow https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report for details :)

pcbroch, 2016-12-09:
Same here, tried a few cases and it works as expected.

foosel, 2016-12-09:
That saga continues in foosel/OctoPrint#1637 :)

closed 2016-12-09 by foosel