guysoft/OctoPi #283: Raspberry Pi 3 getting extremely hot after upgrade to 1.3.0

zolan, 2016-12-12:
I'm not sure if others are experiencing this or not but ever since my upgrade to 1.3.0 my RPi has been getting ridiculously hot.  I didn't add/remove any addons.  The only thing I did was upgrade from 1.2.16 to 1.3.0

If I end up being the only one this is happening to, please feel free to close this issue but it seems like a heck of a coincidence.

This Pi does nothing but run octoprint.

guysoft, 2016-12-12:
Can you post an output or name of process from top that is taking up the CPU?

zolan, 2016-12-12:
Will do.  I've noticed this seems to happen after a print has finished and shortly after the Pi becomes unreachable.  I will get some information from the next print tonight and try to get you something tomorrow.

seefood, 2016-12-13:
right after the print, the images are rendered into a movie and that's a CPU-heavy task. do you have movie rendering turned on? did you add a heat sink to the board? they are cheap and useful.

Guy - maybe put it under nice or limit it to one core?

nophead, 2016-12-13:
I don't think the rendering should be hobbled. RPI2 and PRI3 need
heatsinks. It is a hardware issue, not software.

zolan, 2016-12-13:
I do have heat sinks, I do not have a fan attached to the Pi but I now have a desk fan blowing on it which may have invalidated this test because for the first time since my upgrade to 1.3.0 my Pi continued working.  Even with my desk fan blowing into the pi, during rendering it appears to have increased 10 degrees C.  I would imagine that would be much higher if there were no fan at all.

I do have timelapse set up and it is probably being worked rather hard as I wanted to see what more frequent pics would look like when rendered but I had been using that change back in 1.2.16.

If this is a case of me simply overworking my pi then I suppose I will make a new case for it that supports a fan and hope that is good enough.  It just seems odd that this cropped up after the update.

guysoft, 2017-09-24:
Closing if no activity. Seems to work fine nowdays.

closed 2018-01-04 by guysoft