guysoft/OctoPi #284: OctoPrint does not startup automatically

mklhmnn, 2016-12-14:
First, thanks everyone who provided OctoPrint and OctoPi.

I have installed the latest OctoPi image on a raspberry pi b+ and basically everything works fine. Except of that when I start up the raspberry pi, OctoPrint is not started (I've waited ~5min). No octoprint process it there, no (new) log entry in ~/.octoprint/logs/. I have to login using SSH and perform

  sudo shutdown -r now

to reboot it, then OctoPrint starts fine. Does anyone has an idea what could cause the described behavior. I'm no Linux expert, but I would assume that rebooting the raspberry pi should cause the same behavior as switching it on the first time. Thanks in advance.

guysoft, 2016-12-16:
It sounds like an OctoPrint issue, I suggest you post whats in ``~/.octoprint/logs/`` 
@foosel sorry to bug you, but I can't find how to start octoprint without the service. ANy ideas?

pcbroch, 2016-12-16:
Isn't it:

~/oprint/bin/python ~/OctoPrint/run --daemon start --port 5000 --pid
/tmp/octoprint

foosel, 2016-12-16:
`~/oprint/bin/octoprint`. Or `~/oprint/bin/octoprint --daemon start` etc for daemon mode. But for a regular manual start the first should work just fine. See also [here](https://github.com/foosel/OctoPrint#usage).

Problem described here sounds weird though. The service file really does nothing else than the above. Either it should work, or shouldn't work (e.g. in case of a mangled `config.yaml` file), but a reboot shouldn't be able to change behaviour without touching anything else. If it is however and a manual start of OctoPrint via the above doesn't help in figuring out the issue, I'd try the regular Raspberry Pi troubleshooting of a) reflashing the image and b) hooking up a solid 5V 2.5A micro usb power supply (the red LED on the Pi needs to be lightened up and solid, not flashing or off).

guysoft, 2016-12-16:
yes sounding like a clasic power supply thing

mklhmnn, 2016-12-16:
Thank you for the feedback. I'm using this Raspberry PI with a 2A power supply and it works absolutely stable - no flashing red LED. It even can power the RAMPS 1.4 board, too, if the printer is not directly connected to its power supply. I reckon, otherwise I would have some strange problems while it operates, but once OctoPrint is started, it works perfectly.

I've tried disconnected from the printer - same problem. I've now tried without the Logitech C270 webcam and - voila - OctoPrint starts fine. So it looks like the cam somehow interferes with OctoPrint on the first start.

foosel, 2016-12-16:
I have the same camera. It doesn't interfere for me. The OctoPrint server also doesn't care if there is a camera or not, it doesn't do something like a cam check on boot or something like that. Personally I'd still lean towards power as a potential factor and try with another slightly beefier PSU, just to rule this out. Not because I don't believe your observations regarding the stability of the system, but because I have seen the weirdest things these past couple of years that were caused by the Pi's somewhat suboptimal power management. Things that didn't manifest in any stability issues but only in weird behaviour (like the printer disconnecting or the server not starting), things that I'd otherwise say are a case for the Ghostbusters. Burnt child and all that :)

nophead, 2016-12-16:
>So it looks like the cam somehow interferes with OctoPrint on the first
start.

If it does then the most likely reason is it takes power, so increases the
demands on the PSU.

Have you looked at dmesg and /var/log/messages to see what goes wrong when
it fails to start?

mklhmnn, 2016-12-23:
I've now tried a [better PSU](https://www.amazon.de/gp/product/B01MRHSHA5), but the result remains the same.

pi@octopi:~ $ dmesg | grep octo
[    2.812263] systemd[1]: Set hostname to <octopi>.

mklhmnn, 2016-12-23:
I've compared the output of dmesg after power-on-boot and after a software-reboot. They only differed (beside the times, of course) only in lines like

[    3.560310] systemd[1]: [/lib/systemd/system/dev-mqueue.mount:6] Assignment outside of section. Ignoring.

so I've taken a closer look at this /lib/systemd/system/dev-mqueue.mount. Strangely, it had some random invalid characters in it. Hence I've reinstalled the OctoPi-image, updated to the latest OctoPrint 1.3.0 and now it seems to work fine even with the old 2A PSU. I'm surprised that with this buggy file it worked for so long.

Thank you for all your help. I wish you happy holidays!

foosel, 2017-03-27:
Sounds like this was solved by fixing that corrupted file/reimaging. Closing.

closed 2017-03-27 by foosel

mklhmnn, 2017-03-28:
This issue only had temporarily been resolved by a new image. It occurred again, but I resolve it by connecting the cam later.

guysoft, 2017-03-28:
@mklhmnn that really makes it sound lile a power issue. Make sure your power supply AND power cable are supplying at least 2 amps of power

nophead, 2017-03-28:
Yes a lot, if not most micro USB cables have far too much resistance for powering an RPI.

mklhmnn, 2017-03-28:
I'm using now that [3A power supply](https://www.amazon.de/gp/product/B01MRHSHA5) since January.

mklhmnn, 2017-12-04:
Just for the records: The problem was **not** caused by file corruption or the power supply. In September 2017 I installed OctoPi 0.14 on exactly the same hardware and never had the problem again.

guysoft, 2017-12-05:
AND power cable

John-Mc, 2017-12-05:
@nophead - do you have a source for known good cables? Or better yet, is there some spec we can look for on the cables?

HBomb2003, 2019-01-03:
You need a powered usb to plug the camera into. It is a power issue