guysoft/OctoPi #287: Rainbow screen on 0.13 stable build and md5 incorrect.

akutruff, 2016-12-28:
Spent the last two days trying to get OctoPi to work.  Using 
https://octopi.octoprint.org/latest  which is pointing to 0.13

RPi 2 from this kit - https://www.adafruit.com/product/2896

I verified that the hardware was working correctly by running a fresh NOOBS install of raspian.

The main download's md5 is not correct.  I verified this on OSX and Windows. 

All I get is a rainbow screen and nothing else.

I then tried the nightly builds but they appeared to stop updating in June.  One of them booted, but I spent hours trying to get the wireless to get an ip address so I figured I'd stop gambling on whether it was a stable build.

guysoft, 2016-12-28:
Hey,
I just downloaded and ran and I get
```
md5sum /tmp/octopi-jessie-lite-0.13.0.zip 
d0191958de7ffe0f0a62c9313b7a3fc9  /tmp/octopi-jessie-lite-0.13.0.zip
```
Which is the correct md5sum on https://octopi.octoprint.org/ .
What are you getting? 

The nightly build HTML files was indeed not updating (not sure why no one said so). Its working now:
http://docstech.net/OctoPiMirror/nightly/

So now you can do 2 things:
1. try and download the see if the md5 is right
2. you can try a nightly build

akutruff, 2016-12-28:
MD5 - I was hashing the img, not the zip.  That's my bad.  Sorry about that.

Still getting rainbow screen on 0.13 stable though.

Just installed nightly. Getting through boot.  Network settings still not working though.  From googling this seems to be the most difficult part of OctoPi.

guysoft, 2017-01-01:
Its not difficult, just edit ``octopi-network.txt`` as explained in it.

closed 2017-01-01 by guysoft

akutruff, 2017-01-01:
That part is not difficult.  It just simply didn't work.