guysoft/OctoPi #290: Octopi/print sending code after aborting causing crash!

totalitarian, 2017-01-11:
When I cancel with the printer buttons, everything works as it should.  It homes x/y and lowers the bed.  But if I cancel from octopi, it stops without lowering the bed.  

So I put in gcode in the "cancel" gcode box to lower the bed/home turn off motors etc.  Seemed to work OK, but if I cancel from the LCD screen on the printer, the printer processes the cancel - homes and lowers the bed.  Then is seems Octoprint processes the gcode and tries to lower it again.  The printer doesn't seem to track the Z position is at the bottom after it cancels so thinks it has to lower again causing a crash.

Is this know issue?

nophead, 2017-01-11:
It shouldn't be a problem telling the printer to go to the same Z position twice as long as you use absolute moves. If you switch to relative mode it could be a problem, but most printers have soft limits that would still prevent the Z axis going too far.

Can you post the gcode you are using to lower the bed?

totalitarian, 2017-01-11:
```
G90 ; absolute positioning
M73 P100 ; end build progress
G1 Z150 F1000 ; send Z axis to bottom of machine
M140 S0 T0 ; cool down HBP
M104 S0 T0 ; cool down right extruder
M104 S0 T1 ; cool down left extruder
M127 ; stop blower fan
G162 X Y F3000 ; home XY maximum
M18 ; disable stepper
```

nophead, 2017-01-11:
It could be that the M18 causes it to lose track of its position.

totalitarian, 2017-01-11:
I'm sure it crashes before that line

nophead, 2017-01-11:
Yes but if the first time it gets M18 it loses track of the Z position the next time it runs G1 Z150 F1000 it may crash. So it is before the second M18 but after the first one.

guysoft, 2017-01-11:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

totalitarian, 2017-01-11:
OK I think it'll leave out any z movements and just home x and y

nophead, 2017-01-11:
This is one of the reasons I home Z away from the bed, so at the end of a build I can home Z safely regardless of the state of the machine.

foosel, 2017-03-27:
As already mentioned, wrong bug tracker.

closed 2017-03-27 by foosel