guysoft/OctoPi #292: [Request] Webcam Security

KairuByte, 2017-01-16:
Is it not possible to add security to the webcam aspect? Even simple authentication where it pops up asking for you to authenticate would be preferable to nothing at all. I believe this would be possible with a reverse proxy, however I admit I am not familiar enough with linux to say for certain if this can be accomplished.

Salandora, 2017-01-16:
Well as I know haproxy can make this for your, and @foosel had an experiment with nginx authenticating through octoprint.

So it is possible but needs extra configuration, which for most of the people I think might not be the default situation.

KairuByte, 2017-01-22:
I actually found that and she directed me to the gitter related to it. I'm giving it a shot tonight. It would be interesting to have that as a script as part of octopi to turn it on.

KairuByte, 2017-01-30:
I got the nginx method working pretty nicely. Hopefully it make's its way into OctoPi.

Thanks!

guysoft, 2017-01-30:
Is there a pull request?
I know @foosel mentioned she was thinking about it

KairuByte, 2017-01-30:
Not that I am aware of. She pointed me towards a gist she wrote a few months back and I adapted that into a script that I run after install. I made a few modifications as it seemed she was leaving haproxy in place and installing nginx alongside it.

It works pretty well.

foosel, 2017-01-30:
No PR yet, too much other stuff on my plate and we need to do this in a way that doesn't cause issues for users who actually don't want their webcam protected like this (because I'm absolutely sure if we don't you won't be able to say "webcam" before we have people scream at is for changing something "that was always like this"). The gist currently enforces protection and needs to be expanded to allow the user to choose first.

It's on my list and if you are feeling an urge to do a new release asap that includes that I can move it up in priority, but right now there are more pressing things.

KairuByte, 2017-01-30:
I actually took care of that with a small setting in the plugin for debug purposes.

If I knew more about how this particular project works I would bang it into something fitting of review and make a pull request.

guysoft, 2017-11-21:
This was never fixed, reopened