guysoft/OctoPi #293: Plugin Manager error on latest version

cbrherms, 2017-01-17:
Downloaded and set up OctoPi image last night, once done i noticed that the plugin manager wouldn't connect to the plugin repository, saying that the repository is not available.

Upon looking at the start up logs i've seen the following:

`2017-01-17 10:09:21,314 - octoprint.plugin - ERROR - Error while calling plugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 82, in on_startup
    self._repository_available = self._fetch_repository_from_disk()
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 568, in _fetch_repository_from_disk
    return self._refresh_repository(repo_data=repo_data)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 620, in _refresh_repository
    self._repository_plugins = map(map_repository_entry, repo_data)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 613, in map_repository_entry
    result["is_compatible"]["octoprint"] = self._is_octoprint_compatible(octoprint_version, entry["compatibility"]["octoprint"])
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 632, in _is_octoprint_compatible
    s = next(pkg_resources.parse_requirements("OctoPrint" + octo_compat))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources.py", line 2697, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ("Expected ',' or end-of-list in", 'OctoPrint>=1.2.6+', 'at', '+')`

Cyberman-tM, 2017-01-18:
Same here I think. I tried updating to 1.3.0 with out success.
Which Raspi do you have? I've a Raspi 3.

cbrherms, 2017-01-18:
RasPi 3 also

guysoft, 2017-01-18:
cc @foosel  people here, didnt redirect them in time

foosel, 2017-01-19:
Already solved, was an issue with the version string in an updated plugin entry.

closed 2017-01-19 by guysoft