guysoft/OctoPi #294: Suggestion: add "IPQoS 0x00" to "/etc/ssh/sshd_config", Rpi3, WiFi SSH issue

Cyberman-tM, 2017-01-17:
This probably only affects a few, but it's rather annoying. Details here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138631&p=1085534&hilit=qos#p1085534

Short version: Some(?) Raspi 3 stop responding to SSH after the (correct!) password has been entered.
Attaching a keyboard works, so it's just SSH.

Adding the line
**IPQoS 0x00**
to the file
**/etc/ssh/sshd_config**
solves the issue.

I have no idea what IPQoS is, but I do know it solved the issue of not being able to log in remotely at once. (After restarting the ssh daemon or the raspi.)

guysoft, 2017-01-18:
Where exactly do you need to add this? Can you send your ``/etc/ssh/sshd_config``?

Cyberman-tM, 2017-01-19:
I've added it near the bottom, above the last paragraph. Can't give you the details right now, I'm at work, sorry.
I don't think it matters where it's put, but as I said I don't understand why it solves the issue either.)

Cyberman-tM, 2017-01-20:
Finally, here's the sshd_config file content:
```
# Package generated configuration file
# See the sshd_config(5) manpage for details

# What ports, IPs and protocols we listen for
Port 22
# Use these options to restrict which interfaces/protocols sshd will bind to
#ListenAddress ::
#ListenAddress 0.0.0.0
Protocol 2
# HostKeys for protocol version 2
HostKey /etc/ssh/ssh_host_rsa_key
HostKey /etc/ssh/ssh_host_dsa_key
HostKey /etc/ssh/ssh_host_ecdsa_key
HostKey /etc/ssh/ssh_host_ed25519_key
#Privilege Separation is turned on for security
UsePrivilegeSeparation yes

# Lifetime and size of ephemeral version 1 server key
KeyRegenerationInterval 3600
ServerKeyBits 1024

# Logging
SyslogFacility AUTH
LogLevel INFO

# Authentication:
LoginGraceTime 120
PermitRootLogin without-password
StrictModes yes

RSAAuthentication yes
PubkeyAuthentication yes
#AuthorizedKeysFile     %h/.ssh/authorized_keys

# Don't read the user's ~/.rhosts and ~/.shosts files
IgnoreRhosts yes
# For this to work you will also need host keys in /etc/ssh_known_hosts
RhostsRSAAuthentication no
# similar for protocol version 2
HostbasedAuthentication no
# Uncomment if you don't trust ~/.ssh/known_hosts for RhostsRSAAuthentication
#IgnoreUserKnownHosts yes

# To enable empty passwords, change to yes (NOT RECOMMENDED)
PermitEmptyPasswords no

# Change to yes to enable challenge-response passwords (beware issues with
# some PAM modules and threads)
ChallengeResponseAuthentication no

# Change to no to disable tunnelled clear text passwords
#PasswordAuthentication yes

# Kerberos options
#KerberosAuthentication no
#KerberosGetAFSToken no
#KerberosOrLocalPasswd yes
#KerberosTicketCleanup yes

# GSSAPI options
#GSSAPIAuthentication no
#GSSAPICleanupCredentials yes

X11Forwarding yes
X11DisplayOffset 10
PrintMotd no
PrintLastLog yes
TCPKeepAlive yes
#UseLogin no

#MaxStartups 10:30:60
#Banner /etc/issue.net

# Allow client to pass locale environment variables
AcceptEnv LANG LC_*

Subsystem sftp /usr/lib/openssh/sftp-server
IPQoS 0x00
# Set this to 'yes' to enable PAM authentication, account processing,
# and session processing. If this is enabled, PAM authentication will
# be allowed through the ChallengeResponseAuthentication and
# PasswordAuthentication.  Depending on your PAM configuration,
# PAM authentication via ChallengeResponseAuthentication may bypass
# the setting of "PermitRootLogin without-password".
# If you just want the PAM account and session checks to run without
# PAM authentication, then enable this but set PasswordAuthentication
# and ChallengeResponseAuthentication to 'no'.
UsePAM yes
```

guysoft, 2017-01-25:
Added it to nightly build, would be appreciated if you could test it helps

Cyberman-tM, 2017-02-14:
Sorry, I completely forgot about this :-(

If you still need me to test it, I can probably do that tomorrow - although right now I've got problems reaching the server with the nightlies?

guysoft, 2017-02-15:
Ill fix that. Internet is affected by storms here

guysoft, 2017-02-15:
Fixed the server, try now

guysoft, 2017-02-15:
Down again, will fix

guysoft, 2017-02-15:
Fixed, sorry for the mess, local connection was a mess here

timmmmmey, 2017-02-17:
I can confirm the problem and the fix !

closed 2017-02-18 by guysoft

guysoft, 2017-02-18:
@timmmmmey @Cyberman-tM I want to close this too, you didn't need to enable anything, right?
https://github.com/guysoft/OctoPi/issues/286

Cyberman-tM, 2017-02-18:
I'm testing right now - sorry, forgot again - having problems connecting right now - getting "connection refused", but I'm not sure if the problem is the Raspi or my new modem...

I'll keep trying until I figure out what's happening, I'll write again today.

lechu77, 2021-01-27:
Same here using rpi4, fixed adding " IPQoS 0x00" to /etc/ssh/sshd_config

guysoft, 2021-01-27:
@lechu77 What version of OctoPi are you using? because this has been fixed in later versions here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L77

morezco, 2023-03-30:
This works! Many thanks!

bittorf, 2024-08-05:
It does not work with the recent OpenSSH update anymore,  
but instead of using ```0x00``` one can use ```throughput```, e.g.:

```
user@host:~$ cat ~/.ssh/config
Host *
  HostKeyAlgorithms=+ssh-rsa
  PubkeyAcceptedKeyTypes=+ssh-rsa
  IPQoS throughput
```

guysoft, 2024-08-05:
@bittorf if this has changed then it needs to be updated here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L130
Is there some documentation to why/how it was changed?

commit e3aa517b (2017-02-18): Fixes #294