guysoft/OctoPi #295: [Bug?]Date/Time is way off on new install. NTP shows INIT

Cyberman-tM, 2017-01-20:
Raspi 3, fresh install. Tried to update OctoPrint, failed.
I noticed the date is wrong - about a year back - and wouldn't update.
I set the timezone/location via raspi-config, but that didn't change.
I restarted the NTP service, no effect.

`sudo ntpd -gq` just hangs apparently

I get this all the time:
`
pi@AuxEquip9:~ $ sudo ntpq -p
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
 service02.vie.x .INIT.          16 u    -   64    0    0.000    0.000   0.000
 boxi.trexler.at .INIT.          16 u    -   64    0    0.000    0.000   0.000
 time2.mediainve .INIT.          16 u    -   64    0    0.000    0.000   0.000
 server.samoylyk .INIT.          16 u    -   64    0    0.000    0.000   0.000
`
(The servers change, though. It's not always the same. Not sure why, I did set the location twice, just to try again.)

I don't *think* it's a firewall issue, as the old Raspi (1) didn't seem to have that issue - at least I never noticed anything.
Nor did I find any settings on the firewall that should prevent NTP from working, unless it's on the Raspi itself which I doubt.

guysoft, 2017-01-23:
Its technically a raspbian issue.
I can import this change from FullPageOS to fix it:
https://github.com/guysoft/FullPageOS/commit/651824489f1816779180b3904556e0f2cc32885b

foosel, 2017-03-27:
Looks like what was fixed with #327

closed 2017-03-28 by guysoft