guysoft/OctoPi #296: [request] Keep temperature when cancelling print job

henrikkoch, 2017-01-22:
I would like a feature that enables the printer to keep the current settings of bed and nozzle temp. even though you cancel a print job. In that way you can start another job right after if you by any reason had to stop the current job. In the current setup depending of how fast you are on your fingers in between two jobs, then you have to way for mostly the bed going up in temp again. Thats a little "crazy" just to wait several minutes if you just want to continue with a new job.

guysoft, 2017-01-23:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

foosel, 2017-03-27:
As already mentioned, wrong bug tracker.

closed 2017-03-27 by foosel