guysoft/OctoPi #297: [Request] User programmed home position

henrikkoch, 2017-01-22:
I would like to have button in the Temperature page and Control page which can be maybe combined button to both Cancel the current print and go to a user predefined position that fits for your special printer. This should also include both a X,Y,Z position.

When using the home today - it moves the bed back so that you are not able to clean it with a spatula as the X axis is in front of these. Therefore I would like to be able to push a button which then moves the print head to the left and the Y-plane( bed) toward the panel (i have prusa i3 MK2). And also to be able to clean the nozzle the Z up for instance in 4 cm. Then I can easily clean both the nozzle and the bed immateriality after stopping the current print job (if something goes wrong - and that happens from time to time) :-)

guysoft, 2017-01-23:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

foosel, 2017-03-27:
As already mentioned, wrong bug tracker.

closed 2017-03-27 by foosel