guysoft/OctoPi #299: Save connection settings

henrikkoch, 2017-01-22:
Why is the "Save connection setting" hidden when the connection is operational. Its in that moment it could be a god way of signaling that the current setting which according to the printer state "Operational" can now be saved with good confidence to that the current connection parameters is working and therefore can be used the next time it connects.

It looks like this setting: Save connection setting" is not being used for anything as it use a lot of time to communicate with the printer also next time it connects.

NB: I'm using the Prusa i3 MK2

guysoft, 2017-01-23:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

foosel, 2017-03-27:
As already mentioned, wrong bug tracker.

closed 2017-03-27 by foosel