guysoft/OctoPi #300: Possible programmable toggle buttons in UI

henrikkoch, 2017-01-22:
I world like to have some uset programmable buttons (toggle) that can be controller from the webpage and that the user can use to map to a I/O pin. This could make it possible to turn light on/off and external fan on/off

guysoft, 2017-01-23:
This is an OctoPrint issue, not an OctoPi, issue, please post it to https://github.com/foosel/OctoPrint/issues/

KairuByte, 2017-01-31:
This can already be accomplished with the System Command Editor plugin to a certain extent. http://plugins.octoprint.org/plugins/systemcommandeditor/

Alternatively you could use Custom Controls if you wanted it in the control section. http://plugins.octoprint.org/plugins/customControl/

foosel, 2017-03-27:
As already mentioned, wrong bug tracker and already possible.

closed 2017-03-27 by foosel