guysoft/OctoPi #301: Network with ESSID, Username, and Password

pepaulson, 2017-01-25:
I am in progress of integrating OctoPi into our University additive manufacturing laboratory. 
I have setup everything quite well with a wired connection, but am running into obstacles when I attempt to connect via WLAN. 
Our school uses a WPA2 network, which requires the ESSID, personal username, and personal password for connection.
The [/boot/octo-network.txt file](https://github.com/guysoft/OctoPi/blob/e12ae0111cc05d9674005ef8a6b1f3e7ac833438/src/filesystem/boot/octopi-network.txt) only has SSID and password under WPA2, and is missing the username input field. Would it be possible to add the third field in the center, and what would the call-out be? I have tried this with 'wpa-usr "username"' without success.
I spent time on the OctoPi respository and google unable to find any solutions, so any ideas are appreciated... I'm hoping it is a simple misunderstanding on my part, Thanks!

guysoft, 2017-01-25:
Hey, you probably need to set it via ``/etc/wpa_supplicant.conf``
There might be a way in octopi-network.txt, but I don't know of it.
See if this works: /etc/wpa_supplicant.conf:

guysoft, 2017-03-30:
update: there is now a possibility to edit wpa_supplicant.conf in the pre-released 0.14.0 nightly builds.

APAeroDan, 2017-08-05:
I have a similar situation, but cannot get the pi in the network due to a white list security system.  Is it possible to set the octopi up as a hotspot or ad-hoc network?

guysoft, 2017-08-05:
@APAeroDan If its a pi3 there is but no documentation. Also if you want to continue this please open a separate issue because this issue is about wifi with a usernane and password.

guysoft, 2017-11-01:
Closing if no reply

guysoft, 2017-11-04:
Er - noticed that these guys posted a possible solution in a fork:
https://github.com/MSUMakerspace/OctoPi#msu-makerspace

Again, closing if no reply