guysoft/OctoPi #302: Upload to SD Greyed Out

barnold96, 2017-01-28:
This is a new install using the 0.13.0 image on a Raspberry Pi 3B. It is attached to an Anet A8 (Prusa i3 clone) 3D printer. I attach to the RPi via its built in wifi on my home wifi system. I use Firefox for my browser on Win10. The interface works great to "Upload Locally" and run a print job. When I try to drag and drop a gcode file to "Upload to SD", it says "SD Card Not Initialized". I have formatted the SD card in the Anet board using SD Formatter and it shows up normal on File Explorer.

Also, it appears that the monitoring and control functions work fine with the exception of the "Motors Off" button. I'll submit that as a seperate issue.

KairuByte, 2017-02-03:
I replied to your OctoPrint issue as well I believe, but these are likely both issues with your firmware. The default firmware does not play nice with being controlled by a PC, you need to replace it with something like Skynet3D.

foosel, 2017-03-27:
As mentioned, not an issue with OctoPrint but with your firmware. And in any case this is the wrong bug tracker (because it's the one for Octo**Pi**).

closed 2017-03-27 by foosel