guysoft/OctoPi #304: [Request] "Public friendly" error pages

KairuByte, 2017-01-30:
It would be nice if there were a set of "public friendly" error pages on top of the descriptive error pages that currently exist. This would be useful for servers that are fully configured and working, and also facing WAN. Users could see that there is a problem, but won't see the technical jargon that a lot of people wouldn't understand.

This might be more useful once user accounts are fully fleshed out.

guysoft, 2017-01-30:
can you describe what kind of errors you expect to see? There are already
some messages and i want to know what you expect to see when.

KairuByte, 2017-01-30:
Ah, I should have been more specific. 

I'm meaning pages that say "Octoprint is down" _without_ going into the troubleshooting steps. Personally once I have set up my server I don't need instructions on the error pages I receive, and users that share access don't gain anything by seeing it either as they don't have access to SSH. 

I actually like both of the current errors (no webcam/Octoprint).

guysoft, 2017-09-24:
Closing if no activity, more related to OctoPrint or OctoPi.

closed 2017-11-01 by guysoft