guysoft/OctoPi #305: Unable to do a build of OctoPi

mgriesi, 2017-02-02:
Hi - I'm trying to do a build of OctoPi following your steps.

sudo apt-get install gawk util-linux realpath qemu-user-static git
git clone https://github.com/guysoft/OctoPi.git
cd OctoPi/src/image
curl -J -O -L  http://downloads.raspberrypi.org/raspbian_latest
cd ..
sudo modprobe loop
sudo bash -x ./build

I started with a clean image of the latest jessie on a Pi3, did all the update/upgrades and followed your steps.  Each time I run sudo bash -x ./build it fails.  I noticed it wasn't unzipping the image so I manually did that but still same issue.  I opened all permissions under the OctoPi directory but that didn't help.  There is more than enough space on the SD.  Attached is a session log.  Any thoughts?
Thanks!
[sessionlog.txt](https://github.com/guysoft/OctoPi/files/749192/sessionlog.txt)

guysoft, 2017-02-02:
Looks like its not possible to build OctoPi on a raspberrypi distro without
using the vagrant build method:
https://github.com/guysoft/OctoPi/blob/devel/README.rst#building-using-vagrant

If you want to build using a raspberrypi, you might need to switch to a
vagrant image running ubuntu for arm. Otherwise it means building is
limited to x86 machines.

mgriesi, 2017-02-03:
Thanks.  I'll give that a try and let you know.

mgriesi, 2017-02-03:
It was user error.  While I thought I checked, the zip file name was not "*-raspbian.zip".  I guess that shows when you look at something so many times, you just don't see it.  At any rate, renamed the zip file and your initial steps worked.  Thanks!

closed 2017-02-18 by guysoft